microsoft/Azure-Kinect-Sensor-SDK #1257: Unity body tracking sample crash when it quit

drumath2237, 2020-06-21:
**Describe the bug**

This issue is about [microsoft/Azure-Kinect-Samples](https://github.com/microsoft/Azure-Kinect-Samples) repository.
When I tried to quit the [sample_unity_bodytracking](https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/sample_unity_bodytracking) application, it crashes and doesn’t go successfully.
It seems to me that the action of quitting the application before the background process is completed is the cause of this problem.

Here's a part of code in `Assets/Scripts/main.cs`.

```cs
void OnDestroy()
{
    // Stop background threads.
    if (m_backgroundDataProvider != null)
    {
        m_backgroundDataProvider.StopClientThread();
    }
}
```

In this code, m_backgroundDataProvider's background process assumed to stop, 
but in fact, before the background process stops completely, the attached gameObject got destroyed, therefore the application crashed.
This problem will not occur in playing in editor.
I consider this problem serious, because if a background process remained in computer, sometimes an error occurs when you want to use Azure Kinect next time.

In my forked branch [users/drumath2237/build-crash-fix](https://github.com/drumath2237/Azure-Kinect-Samples/tree/users/drumath2237/build-crash-fix), I tried to solve this problem.
First, I fixed `SkeletalTrackingProvider`'s IsRuuning property to change state when process stoped.
And then, in `main.cs` I change the code to wait process state change to stop. Here's my code.

```cs
void OnDestroy()
{
    // Stop background threads.
    if (m_backgroundDataProvider != null)
    {
        m_backgroundDataProvider.StopClientThread();
        while (m_backgroundDataProvider.IsRunning)
        {
            ;
        }
    }
}
```
If you don't mind, please check it out.

**To Reproduce**

1. Build this project for Windows Standalone
2. Copy DLLs in `sample_unity_bodytracking/' directory to root of build directory.
3. Run `UnityCSharp.exe` and push quit button.

**Expected behavior**

Quit the application correctly.

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: Windows 18363.900
- Azure Kinect SDK version: 1.4.0
 - Body Tracking SDK Version: 1.0.1
 - Firmware version:  Device "000664501412"

**Additional context**

ARportals, 2020-06-22:
It might be happening in the Editor too. I'm having an issue where when going from Play Mode to Editor Mode, a background process does not stop properly. Not sure if it's the same issue mentioned here but I do get an error when reconnecting to the sensor:

```
AzureKinectException: Not all native allocations have been freed before managed shutdown
Microsoft.Azure.Kinect.Sensor.Allocator.ApplicationExit (System.Object sender, System.EventArgs e) (at D:/a/1/s/extern/Azure-Kinect-Sensor-SDK/src/csharp/SDK/Allocator.cs:393)
System.AppDomain.DoDomainUnload () (at <437ba245d8404784b9fbab9b439ac908>:0)

D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\csharp\SDK\Device.cs:310 
  at Portal.KinectAzure.SkeletalTrackingProvider.RunBackgroundThreadAsync (System.Int32 id) [0x001ea] in E:\Portal\Unity\BodyTracking\Assets\Scripts\SkeletalTrackingProvider.cs:65 
UnityEngine.Debug:LogError(Object)
Portal.KinectAzure.SkeletalTrackingProvider:RunBackgroundThreadAsync(Int32) (at Assets/Scripts/SkeletalTrackingProvider.cs:134)
Portal.KinectAzure.<>c__DisplayClass8_0:<StartClientThread>b__0() (at Assets/Scripts/BackgroundDataProvider.cs:15)
System.Threading._ThreadPoolWaitCallback:PerformWaitCallback()

D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\csharp\SDK\Device.cs:188 
  at Portal.KinectAzure.SkeletalTrackingProvider.RunBackgroundThreadAsync (System.Int32 id) [0x00050] in E:\Portal\Unity\BodyTracking\Assets\Scripts\SkeletalTrackingProvider.cs:45 
UnityEngine.Debug:LogError(Object)
Portal.KinectAzure.SkeletalTrackingProvider:RunBackgroundThreadAsync(Int32) (at Assets/Scripts/SkeletalTrackingProvider.cs:134)
Portal.KinectAzure.<>c__DisplayClass8_0:<StartClientThread>b__0() (at Assets/Scripts/BackgroundDataProvider.cs:15)
System.Threading._ThreadPoolWaitCallback:PerformWaitCallback()
```

anantley, 2020-06-24:
@drumath2237 

Hi I think I know what the problem is.  OnDestroy doesn't end the application in the standalone.  So just change OnDestroy to OnApplicationQuit() in your main class.  That is working for me.  I will merge the fix shortly.

closed 2020-07-10 by qm13