microsoft/Azure-Kinect-Sensor-SDK #1258: Can't Access Azure Kinect via k4aviewer in Docker Container

msr-peng, 2020-06-24:
**Describe the bug**

**To Reproduce**

I built a docker image based on Ubuntu 18.04 and OpenGL, and installed `k4a-tools`, `libk4a1.4-dev`, which should satisfy all prerequisites to start Azure Kinect. However, when I launch `k4aviewer`, the terminal will return an error like:
 **`[error] [t=33] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (370): find_libusb_device(). Unable to open LIBUSB at index 0`**
[image]
     
Here was what showed in `k4aview` GUI:
[image]
     
After I clicked the `Dismiss` button, this would show:
[image]

However, if I run `k4aviewer` outside the docker container, it can access Azure Kinect and work perfect. I really need Azure Kinect to run in docker container which isolated from my laptop environment. Anyone has idea that how to make the camera work in docker environment?

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04
 - SDK Version:  1.4
 - Firmware version:

wes-b, 2020-06-24:
These errors appear when the K4aViewer can not find a Kinect connected to the PC. Ensure a device is connected to avoid this error and ensure you have [rules.d](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#linux-device-setup) updated.

msr-peng, 2020-06-24:
> These errors appear when the K4aViewer can not find a Kinect connected to the PC. Ensure a device is connected to avoid this error and ensure you have [rules.d](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#linux-device-setup) updated.

Hey @wes-b,

Thanks for replying. I can definitely make sure that Azure Kinect connect to my PC. Here is what showed after run `lsusb` in docker container:
[image]

It's exactly the same as what showed in my host machine.

Actually I also tried simultaneously open 2 `k4aviewer`, one from docker container, one from my host machine. Azure Kinect would work on the host machine, but would report the error like what I said above in docker container.

msr-peng, 2020-06-24:
> These errors appear when the K4aViewer can not find a Kinect connected to the PC. Ensure a device is connected to avoid this error and ensure you have [rules.d](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#linux-device-setup) updated.

I just made the docker container successfully connect to Azure Kinect. To make it work, I should add `--privileged` to give extended privileges to the container when run docker image, like `sudo docker run --privileged ...`

However, the `k4aviewer` GUI still said it can not initialize audio backend (like the image I showed above).

Here is the GUI without audio parts in docker container:
[image]

It seems that the docker container may miss some software support (like audio driver) to run Azure Kinect audio parts. Do you got any idea about that?

closed 2020-07-16 by qm13

Gabry993, 2022-12-06:
Hi, @msr-peng did you manage to find a solution for microphone/audio part?

Chesterguan, 2023-09-17:
Hi Guys, I tried to simulate your jobs with a basic Ubuntu 18.04 image and failed to run `k4aviewer` with the error:
**X11: The DISPLAY environment variable is missing**, 

I tried to set DISPLAY variable with 0.0 and 1.0,  but neither works with below error: 
**X11: Failed to open display <DISPLAY value>**

Any comments will be helpful!! And also if you could kindly share your docker image, that will be best!

k1915361, 2024-02-20:
What worked for me.
Docker container: Debian GNU/Linux 11

```sh
xhost local:docker # or :root
lsusb | grep 097 # must see 5 devices in host and in container
```

docker run options (thanks to [materdd's comment](https://github.com/isl-org/Open3D/issues/5126#issuecomment-1279706921)):

```sh
--gpus 'all,"capabilities=compute,utility,graphics"' # instead of --gpus all 
--privileged 
--volume="/dev:/dev" # optional for USB devices
-e DISPLAY=$DISPLAY
-v /tmp/.X11-unix:/tmp/.X11-unix
```

Test viewer. 

```sh
k4aviewer # must work in host computer: cameras (infrared, depth, color), IMU data, Microphone data
DISPLAY=:1 k4aviewer # in container - viewer opens up in host GUI (without microphone working)
```