microsoft/Azure-Kinect-Sensor-SDK #1260: Device unavailable on Fedora Linux

gaborpapp, 2020-06-29:
**Describe the bug**

Trying to access an Azure Kinect device on Fedora 31/32 on two different computers. `K4aviewer` reports that libusb devices are all unavailable. `K4aviewer` runs, but it is unable to connect to a device, `AzureKinectFirmwareTool -l` displays an error, but also lists the connected device.

**To Reproduce**

1. Install binary packages or build from git
2. Copy `99-k4a.rules` to `/etc/udev/rules.d`
3. Try running `k4aviewer` or `AzureKinectFirmwareTool -l`
4. Error

```
== Azure Kinect DK Firmware Tool == 
[2020-06-29 18:09:02.714] [critical] [t=58677] .../Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (365): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2020-06-29 18:09:02.714] [error] [t=58677] .../Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
Found 1 connected devices:
1: Device "000262493612"
```

```
[2020-06-29 18:23:41.401] [critical] [t=60288] .../Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (365): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2020-06-29 18:23:41.401] [error] [t=60288] .../Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2020-06-29 18:23:41.401] [error] [t=60288 .../Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
```

**Expected behavior**

To be able to access the device without problems from `k4aviewer` and `AzureKinectFirmwareTool`

**Logs**

dmesg displays some error messages:
```
[22004.194482] usb 1-3.3: New USB device found, idVendor=045e, idProduct=097e, bcdDevice= 1.06
[22004.194487] usb 1-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[22004.194491] usb 1-3.3: Product: Azure Kinect Microphone Array
[22004.194493] usb 1-3.3: Manufacturer: Microsoft Corporation.
[22004.264585] usb 2-3.2: Device not responding to setup address.
[22004.476081] usb 2-3.2: Device not responding to setup address.
[22004.677532] usb 2-3.2: device not accepting address 35, error -71
[22005.505635] usb 2-3.1: new SuperSpeed Gen 1 USB device number 37 using xhci_hcd
[22005.518085] usb 2-3.1: Isoc endpoint with wBytesPerInterval of 1024 in config 1 interface 1 altsetting 4 ep 129: setting to 944
[22005.518405] usb 2-3.1: New USB device found, idVendor=045e, idProduct=097d, bcdDevice= 0.0c
[22005.518406] usb 2-3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[22005.518407] usb 2-3.1: Product: Azure Kinect 4K Camera
[22005.518408] usb 2-3.1: Manufacturer: Microsoft
[22005.518408] usb 2-3.1: SerialNumber: 000262493612
[22005.522217] uvcvideo: Found UVC 1.00 device Azure Kinect 4K Camera (045e:097d)
[22005.522333] uvcvideo: Failed to query (GET_INFO) UVC control 3 on unit 1: -32 (exp. 1).
[22005.524010] uvcvideo: Failed to set UVC probe control : -32 (exp. 26).
[22005.527343] uvcvideo 2-3.1:1.0: Entity type for entity Extension 2 was not initialized!
[22005.527344] uvcvideo 2-3.1:1.0: Entity type for entity Extension 6 was not initialized!
[22005.527344] uvcvideo 2-3.1:1.0: Entity type for entity Extension 5 was not initialized!
[22005.527345] uvcvideo 2-3.1:1.0: Entity type for entity Processing 3 was not initialized!
[22005.527345] uvcvideo 2-3.1:1.0: Entity type for entity Camera 1 was not initialized!
[22005.527413] input: Azure Kinect 4K Camera as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3.1/2-3.1:1.0/input/input47
```

**Desktop (please complete the following information):**
 - OS with Version: Fedora Linux 31 and 32, Kernel 5.6.19
 - SDK Version:  1.4.1
 - Firmware version:   `AzureKinectFirmwareTool.exe -q` only displays errors. If this is relevant I can provide it from a windows machine later.

**Additional context**

Does not work with and without `sudo` either.
Also tried to put `libdepthengine.so.2.0` and/or the `libk4a1.4/` folder on `LD_LIBRARY_PATH`.
As far as I remember everything worked with SDK 1.3.1 on the same machine some time ago.

qm13, 2020-06-29:
@gaborpapp we officially only support Ubuntu Linux 18.04 and have not tested on Fedora Linux. There is not enough info in the logs you posted to identify the cause of the issue. If you could run logging with a more verbose setting and post the logs we may be able to identify the cause.

gaborpapp, 2020-06-29:
@qm13 Thank you. The logs with info level:
```
New logging session started

[2020-06-29 22:24:06.863] [critical] [t=6949] .../Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (365): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2020-06-29 22:24:06.863] [error] [t=6949] .../Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2020-06-29 22:24:06.863] [error] [t=6949] .../Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
[2020-06-29 22:24:11.407] [info] [t=6949] .../Azure-Kinect-Sensor-SDK/src/dynlib/dynlib_linux.c (125): dynlib_find_symbol(). Depth Engine loaded .../usr/lib/x86_64-linux-gnu/libk4a1.4/libdepthengine.so.2.0
[2020-06-29 22:24:11.407] [info] [t=6949] .../Azure-Kinect-Sensor-SDK/src/deloader/deloader.cpp (48): verify_plugin(). Loaded Depth Engine version: 2.1.3
```

wes-b, 2020-06-29:
Well that didn't help.

What does 'lsusb' return? 

```
[2020-06-29 22:24:06.863] [critical] [t=6949] .../Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (365): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
```
This implies its not seeing ANY Azure Kinect USB end point, so you will have to figure out how to make the USB devices visible to LibUsb. You should see something like this: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1258#issuecomment-648930992.

Maybe validate your cables, power supply, etc, by using only the gear that came with the device to validate basic functionality.

gaborpapp, 2020-06-30:
`lsusb` output seems similar, although I'm missing `045e:097c`

```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 005: ID 045e:097d Microsoft Corp. 
Bus 002 Device 002: ID 045e:097a Microsoft Corp. 4-Port USB 3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 1532:023b Razer USA, Ltd Razer Blade
Bus 001 Device 004: ID 13d3:56bd IMC Networks USB Camera
Bus 001 Device 003: ID 8087:0025 Intel Corp. 
Bus 001 Device 007: ID 045e:097e Microsoft Corp. 4-Port USB 2.0 Hub
Bus 001 Device 006: ID 045e:097b Microsoft Corp. 4-Port USB 2.0 Hub
Bus 001 Device 002: ID 046d:c05b Logitech, Inc. M-U0004 810-001317 [B110 Optical USB Mouse]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

gaborpapp, 2020-06-30:
It seems to be working now. It was a power cable issue. Thanks for your help @wes-b @qm13 !

closed 2020-06-30 by gaborpapp