microsoft/Azure-Kinect-Sensor-SDK #1263: Requesting Binaries for 20.04

nicolas-schreiber, 2020-07-02:
Hi,

I am trying to install this SDK on Ubuntu 20.04, but there are not binaries installable through the Repository yet. Building works, but to run it I need the Depth Engine...

qm13, 2020-07-02:
Please up vote the UserVoice entry https://feedback.azure.com/admin/v3/ideas/40368301/

krasin, 2020-08-18:
Just for the record, I tried to vote by the link (thx for providing it!) but it failed with an uncomprehensible error when I tried to log in with my Microsoft account:

```
AADSTS50020: User account 'imkrasin@gmail.com' from identity provider 'live.com' does not exist in tenant 'UserVoice, Inc.' and cannot access the application '91a42e81-999b-4cf1-aa36-bb33f25ff53b'(windowsazure.uservoice.com) in that tenant. The account needs to be added as an external user in the tenant first. Sign out and sign in again with a different Azure Active Directory user account.
```

wes-b, 2020-09-03:
> I know they don't officially support 20.04 but they also don't give any explanation of what changed between 18.04 and 20.04 that would make a difference, so I figure it's fine, right?

There is no expected difference, we don't have automation to test on 20.04 ATM. So we don't release packages we can't test.

mpdroid, 2020-10-16:
These are the steps I followed to install k4a-tools, libk4a and libk4abt on Ubuntu 20.04. The general steps are as outlined in 
https://docs.microsoft.com/en-us/azure/Kinect-dk/sensor-sdk-download, with a couple of hacks to make things work on 20.04.
- use of 18.04 repo, even though OS is 20.04
- installed lower versions of tools and libraries (as latest versions of sensor and body tracker don't seem to be compatible on 20.04) 
```
$ curl -sSL https://packages.microsoft.com/keys/microsoft.asc | sudo apt-key add -
$ sudo apt-add-repository https://packages.microsoft.com/ubuntu/18.04/prod
$ curl -sSL https://packages.microsoft.com/config/ubuntu/18.04/prod.list | sudo tee /etc/apt/sources.list.d/microsoft-prod.list
$ curl -sSL https://packages.microsoft.com/keys/microsoft.asc | sudo apt-key add -
$ sudo apt-get update
$ sudo apt install libk4a1.3-dev
$ sudo apt install libk4abt1.0-dev
$ sudo apt install k4a-tools=1.3.0

````
- Verify sensor library by launching camera viewer
````
$ k4aviewer
````

- Clone and build [Azure Kinect Samples](https://github.com/microsoft/Azure-Kinect-Samples).  
- Run `simple_3d_viewer` to verify body tracker works.

martivilella, 2021-01-12:
For the record, another method of getting the latest binaries in Ubuntu 20.04 without adding keys for apt is by simply downloading them manually from Microsoft's package repository and installing them using dpkg (i.e. `dpkg -i k4a-tools_1.4.1_amd64.deb`).
The relevant package repository folder for k4a-tools is [here](https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/k/k4a-tools/) and for libk4a + libk4a-dev [here](https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/).

atinfinity, 2021-01-21:
Do you have a plan to release deb package for Ubuntu 20.04?
I tried to build Azure Kinect SDK from source code.
But, I could not find libdepthengine.so in <https://packages.microsoft.com/ubuntu/20.04/prod/pool/main/>.

matanster, 2021-02-04:
Reading through the thread I'm a little confused on whether the depth engine is or is not available for use on 20.04 as of now. Hopefully a solution / response to @atinfinity's comment would clarify the current state regarding getting to full functionality on 20.04 ...

vinesmsuic, 2021-02-09:
> These are the steps I followed to install k4a-tools, libk4a and libk4abt on Ubuntu 20.04. The general steps are as outlined in
> https://docs.microsoft.com/en-us/azure/Kinect-dk/sensor-sdk-download, [...]

Hi, have you tried using [Azure_Kinect_ROS_Driver](https://github.com/microsoft/Azure_Kinect_ROS_Driver) with k4atools on Ubuntu 20.04 as well? I am getting  `ERROR: cannot launch node of type [robot_state_publisher/state_publisher]:` on whatever script I run after trying to run on ros noetic.

jackjansen, 2021-04-08:
It would be nice if the K4A team could support Ubuntu 20.04 officially. The workaround posted by @vinesmsuic works just fine, for development, but it is a bit of a nuisance (actually more than a bit) to integrate it into our ci/cd setup... And reverting to 18.04 for ci/cd doesn't cut it anymore because of other dependencies where we need versions newer than available on 18.04.

fplk, 2021-04-19:
Fully agree with @jackjansen - 20.04 is the current LTS, most people I know work on it, 18.04 is now 3 years old. It should be ultra-high priority to get this updated. Please try to support 20.04 - this is blocking me and others significantly. It would also be great if you could find a way to support future LTS versions earlier. Thank you in advance.

jackjansen, 2021-04-19:
For people needing a workaround to install K4A in a Docker image I have finally found a workaround that works for me (so far:-), I've posted it in https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1190#issuecomment-822772494

pucciland95, 2021-11-06:
Hello guys, were someone between you guys able to make the body tracking work with version 1.1 and SDK 1.4.1 always in Ubuntu 20.04?
If I launch k4abt_simple_3d_viewer it returns the following error:

`2021-11-06 21:06:15.935753118 [W:onnxruntime:K4ABT_ORT_Environment, cuda_execution_provider.cc:1885 GetCapability] CUDA kernel not found in registries for Op type: Sign node name: Sign
2021-11-06 21:06:15.937735552 [W:onnxruntime:K4ABT_ORT_Environment, fallback_cpu_capability.h:140 GetCpuPreferedNodes] Force fallback to CPU execution for node: Shape__40
*** buffer overflow detected ***: terminated
Aborted (core dumped)
`

kenavolic, 2021-12-01:
This is how we handle the 18.04/20.04 install automatically from the answers given in this thread:

```
hash lsb_release 2>/dev/null || { echo "ERROR: tool lsb_release needed for analysis"; exit 1; }
os=$(lsb_release -cs)

if [[ $os = "focal" ]]; then
  # Special hack for Azure Kinect Framework not available on Ubuntu 20.04
  # Install k4a SDK from their repository.
  # Bit of a hack, see https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1263
  # The simple solution by @vinesmsuic does not seem to work. This uses the manual 
  # solution by @atinfinity
  # Also, https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1190 is part of this solution (needed to non-interactively accept EULA)
  if [[ "$(dpkg -s libk4a1.4-dev | grep Status | awk '{ print $4 }')" != "installed" ]]; then
	  echo "Retrieving azure kinect packages. Be patient..."
	  curl -sSL https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4a1.4/libk4a1.4_1.4.1_amd64.deb > /tmp/libk4a1.4_1.4.1_amd64.deb
	  curl -sSL https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4a1.4-dev/libk4a1.4-dev_1.4.1_amd64.deb > /tmp/libk4a1.4-dev_1.4.1_amd64.deb
	  echo 'libk4a1.4 libk4a1.4/accepted-eula-hash string 0f5d5c5de396e4fee4c0753a21fee0c1ed726cf0316204edda484f08cb266d76' | sudo debconf-set-selections
	  echo 'libk4a1.4 libk4a1.4/accept-eula select true' | sudo debconf-set-selections
	  sudo dpkg -i /tmp/libk4a1.4_1.4.1_amd64.deb
	  sudo dpkg -i /tmp/libk4a1.4-dev_1.4.1_amd64.deb
  fi
else
  # lib kinect
  curl -sSL https://packages.microsoft.com/keys/microsoft.asc | sudo apt-key add -
  sudo apt-get install -y software-properties-common
  sudo apt-add-repository https://packages.microsoft.com/ubuntu/18.04/prod
  sudo apt-get update
  echo 'libk4a1.4 libk4a1.4/accepted-eula-hash string 0f5d5c5de396e4fee4c0753a21fee0c1ed726cf0316204edda484f08cb266d76' | sudo debconf-set-selections
  echo 'libk4a1.4 libk4a1.4/accept-eula select true' | sudo debconf-set-selections
  sudo apt-get install -y --no-install-recommends libk4a1.4-dev
fi
```

apockill, 2021-12-01:
A quick FYI- if you don't want to have to save the hash strings to get around the EULA, I found that this works well in docker builds: 

```
yes yes | DEBIAN_FRONTEND=dialog apt-get install -y /tmp/*.deb
```

nobuyukioishi, 2021-12-07:
Thank you, guys! This thread greatly helped me install this SDK on my Ubuntu 20.04.

Though @vinesmsuic mentioned that "the latest versions of sensor and body tracker don't seem to be compatible", at least the `simple_3d_viewer` demo in Azure Kinect Samples is now working on my environment with the following versions:

k4a-tools 1.4.1
libk4a1.4-dev
libk4a1.4
libk4abt1.1-dev
libk4abt1.1

I hope Microsoft will soon officially support Ubuntu 20.04.

jackjansen, 2022-04-19:
Is there _any_ chance there will be an official apt package for 20.04, ever? @wes-b or anyone else knows?

Ubuntu 22.04 is imminent...

And the workaround with manually downloading the dpkg's works fine, also in my CI/CD setups, but I would like to distribute _my_ software for `apt` too, and the dependencies become pretty much unusable if there isn't an official way to install the k4a packages...

floe, 2022-04-30:
FWIW, I can still install `libk4a*` and friends on 22.04 by just continuing to use the 18.04 repo entry, however, the `k4a-tools` package depends on `libsoundio1` which is no longer available on 22.04. This only works because I upgraded from 20.04 and still had the `libsoundio1` package lying around.

rajkundu, 2022-06-10:
> FWIW, I can still install `libk4a*` and friends on 22.04 by just continuing to use the 18.04 repo entry, however, the `k4a-tools` package depends on `libsoundio1` which is no longer available on [...]

I tried to install the SDK on a 22.04 installation but couldn't get it to work for this reason. Any idea how/where exactly `libsoundio1` is installed on your system? I tried making symlinks to my manually-installed `.so` file for the library, but `k4aviewer` still wouldn't work for me - couldn't find `libsoundio.so.1` despite all of my symlinking.

Any chance you could possibly run `ldd /usr/bin/k4aviewer` and paste the output here? Thank you!

floe, 2022-06-11:
> Any chance you could possibly run `ldd /usr/bin/k4aviewer` and paste the output here? Thank you!
```
[0] $ ldd /usr/bin/k4aviewer
	linux-vdso.so.1 (0x00007ffd259e5000)
	libk4arecord.so.1.4 => /lib/x86_64-linux-gnu/libk4arecord.so.1.4 (0x00007fed55a00000)
	libsoundio.so.1 => /lib/x86_64-linux-gnu/libsoundio.so.1 (0x00007fed559e3000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007fed5595c000)
	libk4a.so.1.4 => /lib/x86_64-linux-gnu/libk4a.so.1.4 (0x00007fed55600000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fed56221000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fed5621a000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007fed554c0000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fed56215000)
	libXrandr.so.2 => /lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007fed55dc9000)
	libXinerama.so.1 => /lib/x86_64-linux-gnu/libXinerama.so.1 (0x00007fed56210000)
	libXxf86vm.so.1 => /lib/x86_64-linux-gnu/libXxf86vm.so.1 (0x00007fed55955000)
	libXcursor.so.1 => /lib/x86_64-linux-gnu/libXcursor.so.1 (0x00007fed55949000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fed55294000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fed551ad000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fed5518d000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fed54f65000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fed56228000)
	libjack.so.0 => /lib/x86_64-linux-gnu/libjack.so.0 (0x00007fed54f10000)
	libpulse.so.0 => /lib/x86_64-linux-gnu/libpulse.so.0 (0x00007fed54ebb000)
	libasound.so.2 => /lib/x86_64-linux-gnu/libasound.so.2 (0x00007fed54db8000)
	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007fed54d00000)
	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007fed54ccc000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fed54ca2000)
	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fed54c78000)
	libXext.so.6 => /lib/x86_64-linux-gnu/libXext.so.6 (0x00007fed55932000)
	libXrender.so.1 => /lib/x86_64-linux-gnu/libXrender.so.1 (0x00007fed54c6b000)
	libXfixes.so.3 => /lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007fed54c63000)
	libdb-5.3.so => /lib/x86_64-linux-gnu/libdb-5.3.so (0x00007fed54ab4000)
	libpulsecommon-15.99.so => /usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-15.99.so (0x00007fed54a2f000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007fed549e1000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007fed549db000)
	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fed549d3000)
	libsndfile.so.1 => /lib/x86_64-linux-gnu/libsndfile.so.1 (0x00007fed54954000)
	libX11-xcb.so.1 => /lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007fed5494f000)
	libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007fed54888000)
	libasyncns.so.0 => /lib/x86_64-linux-gnu/libasyncns.so.0 (0x00007fed54880000)
	libapparmor.so.1 => /lib/x86_64-linux-gnu/libapparmor.so.1 (0x00007fed5486b000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007fed54853000)
	libFLAC.so.8 => /lib/x86_64-linux-gnu/libFLAC.so.8 (0x00007fed54817000)
	libvorbis.so.0 => /lib/x86_64-linux-gnu/libvorbis.so.0 (0x00007fed547ea000)
	libvorbisenc.so.2 => /lib/x86_64-linux-gnu/libvorbisenc.so.2 (0x00007fed5473f000)
	libopus.so.0 => /lib/x86_64-linux-gnu/libopus.so.0 (0x00007fed546e1000)
	libogg.so.0 => /lib/x86_64-linux-gnu/libogg.so.0 (0x00007fed546d6000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007fed546ab000)
	libzstd.so.1 => /lib/x86_64-linux-gnu/libzstd.so.1 (0x00007fed545dc000)
	liblz4.so.1 => /lib/x86_64-linux-gnu/liblz4.so.1 (0x00007fed545ba000)
	libcap.so.2 => /lib/x86_64-linux-gnu/libcap.so.2 (0x00007fed545af000)
	libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007fed54471000)
	libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x00007fed54464000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007fed5443e000)
```

floe, 2022-06-11:
How about giving https://debian.pkgs.org/11/debian-main-amd64/libsoundio1_1.1.0-1_amd64.deb.html a try? Maybe that might help.

samueltorche, 2022-07-25:
Any update on this issue ?
I tried the procedure posted by @mpdroid and @kenavolic but the depth camera is not working, only the RGB.
Anyone had success with making the depth camera work on Ubuntu 20.04 ?

k4a-tools 1.4.1
libk4a1.4-dev
libk4a1.4
libk4abt1.1-dev
libk4abt1.1

rajkundu, 2022-07-26:
This issue is actually a known issue (#387) which occurs without even having any device attached (I have gotten it myself as well). Is your Azure device attached? Also, what color is its LED? If it is orange/amber, it may not be getting enough power, so try using the 5V DC in addition to the USB connection. The LED should be solid white, I believe.

wmcelderry, 2022-07-26:
The popularity tracker/'up vote' topic link above doesn't work.

I think it is now available at:
https://feedback.azure.com/d365community/idea/013e0e1d-f724-ec11-b6e6-000d3a4f0da0

wmcelderry, 2022-08-03:
I got this working on 22.04, thanks to the foresight of the development team:  they have created a docker build environment.

I've used their docker environment on 18.04 to build it, pull out the missing libsoundio1.so for linking purposes (though sound isn't working in k4aviewer on the host system) and run it on the host system.

I've publish the script (WIP) in a dedicated repo [azure_kinect_sdk_scripts](https://github.com/wmcelderry/azure_kinect_sdk_scripts) - hopefully it may help someone.

katiemcb, 2022-08-04:
> > These are the steps I followed to install k4a-tools, libk4a and libk4abt on Ubuntu 20.04. The general steps are as outlined in
> > https://docs.microsoft.com/en-us/azure/Kinect-dk/sensor-sdk-download, with a couple of hacks to make things work on 20.04.
> > 
> > * use of 18.04 repo, even though OS is 20.04
> > * installed lower versions of tools and libraries (as latest versions of sensor and body tracker don't seem to be compatible on 20.04)
> > 
> > ```
> > $ curl -sSL https://packages.microsoft.com/keys/microsoft.asc | sudo apt-key add -
> > $ sudo apt-add-repository https://packages.microsoft.com/ubuntu/18.04/prod
> > $ curl -sSL https://packages.microsoft.com/config/ubuntu/18.04/prod.list | sudo tee /etc/apt/sources.list.d/microsoft-prod.list
> > $ curl -sSL https://packages.microsoft.com/keys/microsoft.asc | sudo apt-key add -
> > $ sudo apt-get update
> > $ sudo apt install libk4a1.3-dev
> > $ sudo apt install libk4abt1.0-dev
> > $ sudo apt install k4a-tools=1.3.0
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > * Verify sensor library by launching camera viewer
> > 
> > ```
> > $ k4aviewer
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 
> > * Clone and build [Azure Kinect Samples](https://github.com/microsoft/Azure-Kinect-Samples).
> > * Run `simple_3d_viewer` to verify body tracker works.
> 
> Hi, have you tried using [Azure_Kinect_ROS_Driver](https://github.com/microsoft/Azure_Kinect_ROS_Driver) with k4atools on Ubuntu 20.04 as well? I am getting `ERROR: cannot launch node of type [robot_state_publisher/state_publisher]:` on whatever script I run after trying to run on ros noetic.
> 
> I wonder if Azure Kinect really supports Ubuntu 20.04 with ROS driver.

I am following the steps above, when I get to this step: 
sudo apt-add-repository https://packages.microsoft.com/ubuntu/18.04/prod

I am getting the following error:
Get:1 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Ign:2 https://packages.microsoft.com/ubuntu/18.04/prod focal InRelease                                        
Err:3 https://packages.microsoft.com/ubuntu/18.04/prod focal Release                                          
  404  Not Found [IP: 13.79.133.120 443]
Hit:4 http://ie.archive.ubuntu.com/ubuntu focal InRelease                               
Hit:5 http://ie.archive.ubuntu.com/ubuntu focal-updates InRelease
Get:6 http://security.ubuntu.com/ubuntu focal-security/main amd64 DEP-11 Metadata [40.7 kB]
Hit:7 http://ie.archive.ubuntu.com/ubuntu focal-backports InRelease
Get:8 http://security.ubuntu.com/ubuntu focal-security/universe amd64 DEP-11 Metadata [66.5 kB]
Get:9 http://security.ubuntu.com/ubuntu focal-security/multiverse amd64 DEP-11 Metadata [2,464 B]
Reading package lists... Done               
E: The repository 'https://packages.microsoft.com/ubuntu/18.04/prod focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

I am currently using Ubuntu 20.04, can anyone guild me on how I can resolve this error?

MightyMirko, 2023-01-31:
> 

`$ curl -sSL https://packages.microsoft.com/keys/microsoft.asc | sudo apt-key add -
$ sudo apt-add-repository https://packages.microsoft.com/ubuntu/18.04/prod
$ curl -sSL https://packages.microsoft.com/config/ubuntu/18.04/prod.list | sudo tee /etc/apt/sources.list.d/microsoft-prod.list
$ curl -sSL https://packages.microsoft.com/keys/microsoft.asc | sudo apt-key add -
$ sudo apt-get update
$ sudo apt install libk4a1.3-dev
$ sudo apt install libk4abt1.0-dev
$ sudo apt install k4a-tools=1.3.0
`

you have to add the key aswell

juancarlosmiranda, 2023-02-07:
Dear colleagues, I share with you this little guide with automated hacks for Ubuntu 18.04 and Ubuntu 20.04 in shell script, I did this based on my experience with the sensor. Currently testing on Ubuntu 22.04. I hope this can be helpful. Azure Kinect camera setup (automated scripts for Linux and Windows) https://github.com/juancarlosmiranda/azure_kinect_notes

mindbeast, 2023-06-11:
Hey microsoft. Please keep supporting the SDK or just open source the libdepthengine.so.2.0 so that someone else can do it for you.

graysuki, 2023-07-12:
> This issue is actually a known issue (#387) which occurs without even having any device attached (I have gotten it myself as well). Is your Azure device attached? Also, what color is its LED? If it [...]

The led on my device is solid white, but my situation is the same as above, he can't detect the device

geyan21, 2023-09-05:
> These are the steps I followed to install k4a-tools, libk4a and libk4abt on Ubuntu 20.04. The general steps are as outlined in https://docs.microsoft.com/en-us/azure/Kinect-dk/sensor-sdk-download, [...]

it works!