microsoft/Azure-Kinect-Sensor-SDK #1264: Cannot record Azure Kinect sensor streams to a file

ykesmis, 2020-07-02:
I cannot create a file for the sensor streams recording.

I enter the following command:
 C:\Program Files\Azure Kinect SDK v1.4.0\tools>k4arecorder.exe -l 5 %TEMP%\output.mkv

I expect to create an output file but I get the following errors:

[2020-07-02 22:14:14.740] [critical] [t=20412] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (366): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2020-07-02 22:14:14.750] [error] [t=20412] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth_mcu\depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2020-07-02 22:14:14.751] [error] [t=20412] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
Runtime error: k4a_device_open() failed
[2020-07-02 22:14:14.752] [error] [t=20412] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (53): k4a_device_t_get_context(). Invalid k4a_device_t 0000000000000000
[2020-07-02 22:14:14.754] [error] [t=20412] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (959): Invalid argument to k4a_device_get_serialnum(). device_handle (0000000000000000) is not a valid handle of type k4a_device_t
Runtime error: k4a_device_get_serialnum(device, serial_number_buffer, &serial_number_buffer_size) returned 1
[2020-07-02 22:14:14.756] [error] [t=20412] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (53): k4a_device_t_get_context(). Invalid k4a_device_t 0000000000000000
[2020-07-02 22:14:14.770] [error] [t=20412] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (201): Invalid argument to k4a_device_close(). device_handle (0000000000000000) is not a valid handle of type k4a_device_t

 - **OS with Version**: Windows 10
 - **SDK Version:**  1.4.0
 - **Firmware version:** RGB camera firmware: 1.6.102 Depth camera firmware: 1.6.75 Depth config file: 6109.7 Audio firmware: 1.6.14

qm13, 2020-07-06:
> [2020-07-02 22:14:14.740] [critical] [t=20412] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (366): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?

This implies the device is already in use. Do you have the k4aviewer or sample app or your app running?

ykesmis, 2020-07-08:
Thanks a lot for your hint! The k4aviewer was indeed running. I tried it again and it did work this time.

closed 2020-07-08 by ykesmis