microsoft/Azure-Kinect-Sensor-SDK #1265: Kinect doesn't work after operating for a period of time

qwe953751, 2020-07-03:
The Kinect Viewer is work normally.
I embedded kinect on my other development system using Miscrosoft.Azure.Kinect.Sensor kit Ver1.2

My pc equipment:
CPU: I7
OS: Windows10, x64

I encountered a problem, and the following error occurred occasionally after I operated for a period of time. As attached:
[image]
[image]

I want to know how to solve this problem.
Now, i update Miscrosoft.Azure.Kinect.Sensor kit to version 1.4.  If it still occurred, i will update to here.

wes-b, 2020-07-06:
For some reason the depth camera is not working. It might be because of your USB controller.

qm13, 2020-07-06:
Please use the issue template next time.
Is your Azure Kinect on the latest firmware? How long does it run before failing? What USB controller is being used? What host platform? Can you provide repro steps? Can you share logs?

closed 2020-08-07 by qm13