microsoft/Azure-Kinect-Sensor-SDK #1266: wrong log level for replace_sample(). capturesync_drop -- misclassification, spamming, etc.

diablodale, 2020-07-03:
The k4a_log_level_t of the below code is incorrectly set to `error`. This state in the code is a quite common natural state (up to 30 times per second) in which the underlying subsystem is managing its queue and keeping frames fresh. Nothing wrong, erroneous, dangerous, or warnful about having fresh frames.

Therefore, `error` is not appropriate and leads to spamming the console/message handler and making it exceedingly difficult to find the actual errors within the spam.

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/52cfb4a87639b3db9cb176b8b9ac814e2b5a4900/src/capturesync/capturesync.c#L137-L142

My preference is `K4A_LOG_LEVEL_TRACE`
At worst, it should be `K4A_LOG_LEVEL_INFO`

That it is *not* an error is also supported by @wes-b when they write "...There are other warnings generated..." at https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1112#issuecomment-596688870

### Fix

Trivial. Change `LOG_ERROR` to `LOG_TRACE`

wes-b, 2020-07-06:
Under normal operation this message will never fire. So this is an error message.

We talked about this, and I am leaning toward not fixing this. I do admit when it fires, it is spammy, however it does indicate a system issue. 

The other issue, is that users tend to copy and past default logs into new github issues. If we lower the level of this message, then users by default will get less useful logs. In turn our triage process will be more drawn out by the fact that we have to ask for more detailed logs.

If the spammyness is causing issues, we can consider generating the log message only 1 time, and then disabling the message. But then the sensor will be running and it will appear that no data is arriving.

diablodale, 2020-07-08:
I woke yesterday and read your reply above. Then started on a reply with some data...but now...those messages are gone. I've tried a day to replicate those messages in the OP. Including intentionally writing bugs using the API. Now...they won't appear. That I can't reproduce them, suggests some failure state (independent of this issue) that existed during the day of the OP. On that day, they appeared consistently many times/second for hours when I used the API. So much that I wrote this issue and custom log handler with a regex. But now...silent.

I caution writing C code to satisfy an issue reporting copy/paste procedure. That would be like turning on [Windows debugging](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/boot-parameters-to-enable-debugging) by default so that debug logs would be always there. But that's not the right approach. Instead, I align with actionable logs targeted for the consumer of those logs with the ability to get more detail for debugging and different consumers of logs.

### Line of Distinction

I think the line that exists in this issue's discussion is a line distinguishing:

1. logs to debug the SDK itself. Microsoft is the primary consumer of these. You express that when you write above "...our triage process will be more drawn out...".
2. logs to debug the user application. Microsoft is not the primary consumer of these. Instead, the primary consumer of these is the developer using the SDK, consultants, ISVs, etc. Naturally, you want to make an SDK that will generate logs that these users of the SDK can use for debugging their custom applications. But Microsoft is not as invested in these.

Is the state that generated the log in the OP a state that is only useful to distinguish for Microsoft (the SDK developer)? That makes it bucket 1 above. Which leads to the question, "Who is the consumer of these logs and what is there need?" For us using this SDK...we are bucket 2. We are not focused on debugging the SDK itself. Instead, we are writing custom applications. Log entries that are for bucket 1 are distracting when they spam overwhelmingly -- debilitating because then we can't see the bucket 2 log entries that we *are* focused on.

In general, I don't want multiple log entries for bucket 1. Why? Because they are not actionable. I can't fix them. They are inside the SDK itself. Of course, I can report issues, and as more detailed info is needed I can provide it. But since I can't fix them myself, such a log entry is noise. (I acknowledge this is not absolutely true...there is a grey area).

You already volunteered an approach of log once and then quiet. That's an interesting approach for bucket 1 SDK debugging log entries. This might mitigate some concerns of your triage efficiency, while also freeing logs of bucket 1 spam.

### Example

While trying to reproduce those spammy OP logs entries, I generated other entries with log levels demonstrating ambiguities in the two buckets. For example, this is a series of log entries that appears if I open a specific Kinect, and then try to open that same Kinect a second time. I'll walk through the entries and apply my thinking from above on them.

```
[2020-07-08 15:33:53.160] [critical] [t=3876] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (366): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?

[2020-07-08 15:33:53.160] [error] [t=3876] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth_mcu\depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()

[2020-07-08 15:33:53.160] [error] [t=3876] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()

Exception thrown at 0x00007FFC5F703E49 in Max.exe: Microsoft C++ exception: k4a::error at memory location 0x00000000007DB8B0.
```

Entry 1 should not be `critical`. This is supported by both [Windows](https://docs.microsoft.com/en-us/previous-versions/office/developer/sharepoint-2010/ff604025(v=office.14)) and Linux. Critical is "your Kinect temp is 1000f and is on fire", "wake up the admin", "the apocalypse is here", . But as you can see...this is just because I called open() twice on the same ordinal of a Kinect. Look at this situation from a "user's" perspective. This is a file in-use/locked scenario. Like trying to open my Word doc and I'm not able to because someone else has it already open. No reasonable person wakes their sysadmin because of this. And therefore, this is not a `critical` condition. It is instead an `error` condition. And it is bucket 1 because this is a USB low-level SDK-only condition. It is not actionable for bucket 2 users of this SDK.

Entry 2 looks also to be an SDK-only low-level bucket 1 condition. I agree it is `error` for a bucket 1 user, but it is spam to bucket 2 users because it is not actionable.

Entry 3 is good. It is actionable. I agree it is `error` and it indicated the user API that failed. I suggest moving the "Is the device being used by another application?" from entry 1 and put it on entry 3. Then this log entry is even more useful.

Entry 4 is the exception throw, vscode seeing it, and my code catches it.

Personally, I would make:

```
Entry 1 = trace
Entry 2 = trace
Entry 3 add suggested sentence
Entry 4 good as is
```

### Surfacing log entries to downstream customers

[... cut ...] y don't want to see the spam from the OP. Its not actionable by them. One approach that I might consider is to surface `critical` errors to these hands-on customers. Though that means that this SDK should only log things critical when it is on fire. Consider in bulk warehouses and monitoring solutions...I might want my Kinects to send status/logs to my centralized systems only when things are `error` or `critical`. Now compare that against the above example. Entries 1 and 2 are not in alignment.

wes-b, 2020-07-10:
You raise great points! We will work to only fire the error 1 time at the existing error level. Still considering if continuing to generate the trace at INFO level is useful, or if we should just remove additional messages,