microsoft/Azure-Kinect-Sensor-SDK #1268: Multi-camera point cloud shader example (feature request / help)

echelon, 2020-07-05:
I was able to port k4aviewer to Rust and have had a great time calling libk4a via the Rust foreign function interface, but I'm struggling to use multiple cameras to build a single cohesive point cloud in OpenGL. My background in OpenGL and GLSL is weak, so that's mostly to blame, but even after reading and making a week of attempts, I'm still spinning my wheels. 

I'm able to sample from two synced cameras and compute the depth from shaders similar to those provided in the k4aviewer code, but I'm not sure how to get the depth and pixel data from multiple cameras into the same point cloud renderer. (I suppose I'll tackle point registration and alignment issues when I get to them; I see there are already several issues on this topic filed that can probably guide me once I get to that point.) 

To clarify, I've gotten the real time 3d point cloud working with one camera. It's almost a carbon copy of the k4aviewer code, except that it's in Rust and I did a few things differently (I rolled my own arcball implementation, etc.) 

I can't figure out how to get two or more cameras to render to the same point cloud. Do the results from the depth compute shaders and the color lookup tables needs to be provided as a uniform? Multiple uniforms? Some type of vertex array buffer? Should use the `gl_VertexID` as an index into the lookup tables? Should the geometry calculation that maps the clouds together happen in C++/Rust or in GLSL? (I'm not at that point yet.) 

I just don't get the structure of what this shader is supposed to look like, probably because the concepts are still a little shaky. 

As an actionable task for the Kinect SDK engineers: I don't think it would be out of scope to add an example of this to the SDK, similar to the provided "Green Screen" code. A lot of people are interested in building this type of system, and this foray into the Kinect world has been my "learn OpenGL" crash course. It would be super awesome if a simple example of multi-camera point cloud rendering (and perhaps point registration) were added to a future release. It's probably not high-priority, but would be extremely beneficial. 

Outside of this request, if someone in the community already has a working example of this they'd be willing to provide below (even just a snippet of the shader and relevant OpenGL calls), I'd really appreciate it. I'd even be willing to pay for a little guidance if you have time. 

Thanks so much!

wes-b, 2020-07-06:
You will want to transform the images from one camera into the coordinate system of the other camera. Our [Green Screen](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/green_screen) example illustrates this.

closed 2020-07-10 by qm13

echelon, 2020-08-14:
Hi @wes-b, thanks for responding! 

I'm sorry I didn't get back in a timely error, but I had to unpack my Kinect sensors from moving them. I also needed to put together a visual explanation of the problem I'm having. 

This isn't an issue with point cloud registration or coordinate systems. It's an OpenGL error arising from naively adapting the k4aviewer code. 

I have two point clouds from two k4a sensors, but only one of them is getting shaded with the webcam color information: https://www.youtube.com/watch?v=5TxeEj4mkFg

Since the other point cloud wasn't getting any color, I shaded it in blue so that it is visible. This has to be due to some error in either the OpenGL code or the GLSL shaders. 

To be clear, this isn't a problem with any of your code. I'm requesting a clearer multi-camera example written in OpenGL (not OpenCV). I'm not interested in solving point cloud registration / coordinate alignment yet. My problems are more fundamental. 

I was able to generalize k4aviwer pretty easily (see [this gist](https://gist.github.com/echelon/ef54135c042a7a5faf2fecded339a04b)), but generalizing the OpenGL to support multiple cameras is tricky. I've been scratching my head for days over this. 

I'd appreciate if you could put this on your roadmap, but I understand if it's outside the scope of your project. 

[image]

[image]