microsoft/Azure-Kinect-Sensor-SDK #1270: Document about the supported body size of Body Tracking

win32-shintaka, 2020-07-06:
Is there any document about the maximum/minimum or recommended body size (especially height) when using Body Tracking with Azure Kinect DK? For instance, a small doll can be recognized as a "body"? How about a baby or a kid?

I think the distance between Azure Kinect DK and the target can be a matter, so I'm expecting there's information about the assumed distance vs. body size.

win32-shintaka, 2020-07-14:
@qm13 , any update on this?

qm13, 2020-07-14:
@win32-shintaka Thanks for resurfacing this issue. Dropped below my radar. Apologies for the delay.

There is no maximum or minimum body size. The algorithm is sensitive to human body proportions rather then size. The BT algorithm has been tested with people from 7 years to 80+ years old. Human body proportions are consistent from ~7 years up. Results for people under 7 years may be inconsistent.

Obviously visual size does matter. An adult 4m from the camera is no different to a child at 1m from the camera.

win32-shintaka, 2020-07-15:
Thank you, @qm13 ! That's enough information for me :)

qm13, 2020-07-15:
Re-opening. Still need to update the Body Tracking SDK documentation.

Chris45215, 2020-08-17:
Can you provide some documentation on the other aspects of body size - specifically weight? A person with a BMI of 19 in form-fitting clothes will look very different from a person with a BMI of 45. What range was the neural network trained with, and at what points is performance expected to drop?
We noticed that it detects people with a missing limb with decent accuracy though not quite as quickly, but the performance difference is usually minor. That aspect works well for us.

qm13, 2020-12-16:
We worked with our ethics team to ensure the training, validation and testing data sets were ethically balanced. This included varying height, body size, and skin tone. The system covers people for 8 years to 99+ years.

closed 2020-12-16 by qm13