microsoft/Azure-Kinect-Sensor-SDK #1271: The master camera acquires the image later than the slave camera, resulting in the acquisition of images completely out of sync

nxe0932, 2020-07-06:
The following picture shows the log, left is Master, and right is the Slave. 
As we can see, the device time of Master captured the first frame shown in the red rectangle in left is extremely later than the device time of Slave captured the first frame shown in the NO.1 green rectangle in right. It is so strange.
[image]
I know Kinect DK use device time is used to synchronization. Even if I aligned the first frame of Master with the third frame of the Slave shown in the NO.2 green rectangle in right, the subsequent device time difference is unstable, some even reaching to 67134 us, it is completely out of sync.
I have even tried on two PCs with different USB controllers and similar results. I have also tried two different sync cables, same result. I have also manually tried different exposure value, same result. And I used the newest 1.4.0 firmware (AzureKinectDK_Fw_1.6.108079014.bin).
Here's the code of controlling the Master and Slave, and any help would be gratefully appreciated!
[image]
[image]

wes-b, 2020-07-06:
Please read through this topic, https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync. A couple pointers:
* You need to start the subordinate camera first.
* You should also use manual exposure. 
* It takes ~10 frames to stabilize timing, especially when using auto exposure.

nxe0932, 2020-07-06:
Thanks to your quick reply. l’m sorry that I didn't make the problem clear.
The above experimental results were based on the topic you mentioned (https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync), including starting the subordinate camera first, and using manual exposure, etc. Unfortunately, as the log shows, the device time (131500) of Master captured the first frame shown in the red rectangle in left is extremely later than the device time (65266) of Slave captured the first frame shown in the NO.1 green rectangle in right. 
Besides, I'm sorry that I don't really understand the meaning of 'It takes ~10 frames to stabilize timing, especially when using auto exposure.'. 
Could you please tell me how to use the time stamp to synchronize the master and slave camera?

wes-b, 2020-07-06:
We don't expect the device timestamp to be stable until ~10 captures after start. Starting the camera, getting the exposure set, and syncing that across the multiple cameras takes time. So sync your device timestamps starting with your 11th capture from master.

With the latest [firmware (1.6.110079014)](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#msis) we expect the captures to be within 100us of each other.

The image of the timing data you shared is too small, so I can't tell if the later frames are synchronizing or not.

nxe0932, 2020-07-07:
Thanks to your quick reply，again. There are the first 20 frames of the Master and the Slave.
**Master**
started_camera_Sys_now[15940219506698095]##
Sys_while_befor[15940219506698843]##frame[1]
Sys_while_after[15940219511785401]-Color_Device_Time[131500]-Frame[1]-Duration_time[0]
Sys_while_befor[15940219512919661]##frame[2]
Sys_while_after[15940219512925772]-Color_Device_Time[164788]-Frame[2]-Duration_time[0]
Sys_while_befor[15940219513657680]##frame[3]
Sys_while_after[15940219513660445]-Color_Device_Time[198133]-Frame[3]-Duration_time[0]
Sys_while_befor[15940219514330574]##frame[4]
Sys_while_after[15940219514333546]-Color_Device_Time[231466]-Frame[4]-Duration_time[0]
Sys_while_befor[15940219515072525]##frame[5]
Sys_while_after[15940219515081463]-Color_Device_Time[264811]-Frame[5]-Duration_time[0]
Sys_while_befor[15940219515850455]##frame[6]
Sys_while_after[15940219515856053]-Color_Device_Time[331466]-Frame[6]-Duration_time[0]
Sys_while_befor[15940219516673699]##frame[7]
Sys_while_after[15940219516678145]-Color_Device_Time[398133]-Frame[7]-Duration_time[0]
Sys_while_befor[15940219517446870]##frame[8]
Sys_while_after[15940219517450291]-Color_Device_Time[498122]-Frame[8]-Duration_time[1]
Sys_while_befor[15940219518271033]##frame[9]
Sys_while_after[15940219518274837]-Color_Device_Time[564800]-Frame[9]-Duration_time[1]
Sys_while_befor[15940219519011551]##frame[10]
Sys_while_after[15940219519015971]-Color_Device_Time[631455]-Frame[10]-Duration_time[1]
Sys_while_befor[15940219519770806]##frame[11]
Sys_while_after[15940219519775857]-Color_Device_Time[731455]-Frame[11]-Duration_time[1]
Sys_while_befor[15940219520470281]##frame[12]
Sys_while_after[15940219520473920]-Color_Device_Time[798133]-Frame[12]-Duration_time[1]
Sys_while_befor[15940219521234338]##frame[13]
Sys_while_after[15940219521244538]-Color_Device_Time[864800]-Frame[13]-Duration_time[1]
Sys_while_befor[15940219521966454]##frame[14]
Sys_while_after[15940219521969291]-Color_Device_Time[931477]-Frame[14]-Duration_time[1]
Sys_while_befor[15940219522593942]##frame[15]
Sys_while_after[15940219522597859]-Color_Device_Time[998122]-Frame[15]-Duration_time[1]
Sys_while_befor[15940219523402589]##frame[16]
Sys_while_after[15940219523408379]-Color_Device_Time[1064800]-Frame[16]-Duration_time[1]
Sys_while_befor[15940219524187100]##frame[17]
Sys_while_after[15940219524191728]-Color_Device_Time[1164788]-Frame[17]-Duration_time[1]
Sys_while_befor[15940219524949536]##frame[18]
Sys_while_after[15940219524956622]-Color_Device_Time[1231455]-Frame[18]-Duration_time[1]
Sys_while_befor[15940219525713145]##frame[19]
Sys_while_after[15940219525717700]-Color_Device_Time[1331466]-Frame[19]-Duration_time[1]
Sys_while_befor[15940219526424580]##frame[20]
Sys_while_after[15940219526427917]-Color_Device_Time[1364800]-Frame[20]-Duration_time[1]

**Slave**
Sys_Start_waiting_now[15940219440820647]
Sys_Start_recive_now[15940219510712033]
Sys_while_befor[15940219510712404]##frame[1]
Sys_while_after[15940219511136059]-Color_Device_Time[65266]-Frame[1]-Duration_time[0]
Sys_while_befor[15940219511777474]##frame[2]
Sys_while_after[15940219511781787]-Color_Device_Time[98600]-Frame[2]-Duration_time[0]
Sys_while_befor[15940219512693290]##frame[3]
Sys_while_after[15940219512701608]-Color_Device_Time[131933]-Frame[3]-Duration_time[0]
Sys_while_befor[15940219513303303]##frame[4]
Sys_while_after[15940219513306005]-Color_Device_Time[165266]-Frame[4]-Duration_time[0]
Sys_while_befor[15940219514109958]##frame[5]
Sys_while_after[15940219514114860]-Color_Device_Time[198600]-Frame[5]-Duration_time[0]
Sys_while_befor[15940219514993676]##frame[6]
Sys_while_after[15940219514997256]-Color_Device_Time[231933]-Frame[6]-Duration_time[0]
Sys_while_befor[15940219515734208]##frame[7]
Sys_while_after[15940219515736965]-Color_Device_Time[298600]-Frame[7]-Duration_time[0]
Sys_while_befor[15940219516574617]##frame[8]
Sys_while_after[15940219516583610]-Color_Device_Time[398600]-Frame[8]-Duration_time[0]
Sys_while_befor[15940219517343227]##frame[9]
Sys_while_after[15940219517345965]-Color_Device_Time[465277]-Frame[9]-Duration_time[0]
Sys_while_befor[15940219517906455]##frame[10]
Sys_while_after[15940219517911156]-Color_Device_Time[531944]-Frame[10]-Duration_time[0]
Sys_while_befor[15940219518691292]##frame[11]
Sys_while_after[15940219518694129]-Color_Device_Time[598600]-Frame[11]-Duration_time[0]
Sys_while_befor[15940219519378710]##frame[12]
Sys_while_after[15940219519385264]-Color_Device_Time[665266]-Frame[12]-Duration_time[0]
Sys_while_befor[15940219520101560]##frame[13]
Sys_while_after[15940219520104554]-Color_Device_Time[765266]-Frame[13]-Duration_time[0]
Sys_while_befor[15940219520843847]##frame[14]
Sys_while_after[15940219520858354]-Color_Device_Time[831933]-Frame[14]-Duration_time[1]
Sys_while_befor[15940219521631408]##frame[15]
Sys_while_after[15940219521634485]-Color_Device_Time[898600]-Frame[15]-Duration_time[1]
Sys_while_befor[15940219522407930]##frame[16]
Sys_while_after[15940219522411019]-Color_Device_Time[998600]-Frame[16]-Duration_time[1]
Sys_while_befor[15940219523149312]##frame[17]
Sys_while_after[15940219523154002]-Color_Device_Time[1065266]-Frame[17]-Duration_time[1]
Sys_while_befor[15940219523789914]##frame[18]
Sys_while_after[15940219523793517]-Color_Device_Time[1131933]-Frame[18]-Duration_time[1]
Sys_while_befor[15940219524514136]##frame[19]
Sys_while_after[15940219524522552]-Color_Device_Time[1198600]-Frame[19]-Duration_time[1]
Sys_while_befor[15940219525327587]##frame[20]
[... cut ...]
We can see from above, there didn't synchronize after 11th frame.  I suspect that I have misunderstood how to synchronize with timestamps.
 A senior told me that _data from different cameras do not arrive at the same time or in turn, generally, each camera needs a background thread to read and cache. Another thread finds synchronous  data from the cache according to their timestamps  ._
Should I write additional code to align the timestamps instead of directly using the timestamps of the frame?
 Could you please  tell me the synchronization principle of multi Kinect DK?

wes-b, 2020-07-07:
You need to use the device timestamp. In this last post is looks like you are using the system timestamp.

closed 2020-08-07 by qm13