microsoft/Azure-Kinect-Sensor-SDK #1272: Depth Distortion Issue

mxf8bv, 2020-07-07:
There is a strong distortion of depth information for flat surfaces that are not parallel to the sensor plane **only if an additional surface is below the object**. See images below. 

Minimal setting:
Condition 1: matte brown cardboard box, surface towards camera is approx 5cmx20cm, distance to camera approx. 1m, angle approx 45 degrees (picture 1 below: side view of point cloud with undistorted front of box)
Condition 2: same but with an additional matte brown cardboard sheet below the box (picture 2 below: same view; flat box front appears concave)

Depth representation in condition 1 is fine. In condition 2, the box surface towards the camera appears strongly concave (approx. 5cm?).

The same effect occurs for different objects on different surfaces (e.g. a table), when the camera is pointing down at around 45 degrees. The central part of a cylinder, for example, appears to be only only 75% of its top and bottom width.

Questions:
1. what is the reason for this effect?
2. can the effect be reduced by different settings?

[image]
[image]

Thanks,
   Mathias

qm13, 2020-07-09:
@mxf8bv could you please also attach color images of the two setups taken with the AK color camera.

mxf8bv, 2020-07-10:
Sure. I attach views of the box frontal to the camera, one time hanging over the table ("noSurface") and once on the table ("withSurface").  
[image]
[image]
[image]
[image]
[image]
[image]

I blurred out parts of our lab in the images.
BTW it's the AK original cardboard box.

qm13, 2020-07-14:
@mxf8bv this looks like an example of severe multi-path. Try placing the box on a surface with low IR reflectivity.

closed 2020-08-07 by qm13