microsoft/Azure-Kinect-Sensor-SDK #1273: IMU accelerometer readings seem incorrect / biased

neilDGD, 2020-07-07:
Hi,

I have been trying to get sensible data from the IMU (first stop positional /acceleration data) and I have observed what I believe to be strange data coming from both the k4a sensors i have tested here. I have observed this both in our own application and the viewer app with IMU sampling enabled.

- Firstly, both the X and Y axis seems to constantly have a ~-0.13m/s^2 offset even when the sensor is stationary, is this normal?
- Secondly and most importantly, when I try and move the sensor in the X axis it never seems to change direction, its always negative. After this confusing me for a while in our application after I integrated IMU sampling, I tried the Viewer app and the same data comes into the viewer app too. I have observed that if I really quickly move the camera in the X direction it gets more 
positive but this far from the use case we need, it would be ideal for our users to use this for camera tracking for SLAM for example and that involves slow camera movements.

I wonder if this could be something to do with calibration on the accelerometer or is there anything else that could be interfering with it? Are there any restrictions to using IMU whilst other streams are running for example, sensor settings?

Also when testing in the viewer app the gyro data, it seems really good reasonable negative to positive ranges on all axis', just the data from the accelerometer _seems_ off.

Any suggestions?

Thanks,

Neil

qm13, 2020-07-09:
@neilDGD if you were able to setup your AK device to be perfectly level in a vibration less environment you would see x=0, y=0, and z=-9.8. Do you think the bias you are reporting is a portion of the gravity vector?
You should be able to rapidly move the device in the x direction and get a positive value. Positive x assuming device is level and facing you occurs when rapidly moving the camera towards you (just tested it on my AK device). It may well be that you have a bad AK.

neilDGD, 2020-07-11:
Quite possible that the building i'm in isn't quite level, given that i'm in Mexico City and we have earthquakes. Haven't got a level rule to check. Still having problems getting good accelerometer values even with removing an initial resting pose bias although removing the bias does visibly give a good resting state with maybe only a tiny bit of drift from the accelerometer. Gyro is another story, getting understandable rotation values but with a lot of error or bias, but this isn't even filtered yet, unknown territory.

neilDGD, 2020-07-11:
I am now getting very small values from the IMU accelerometer, which does not equate to any real world translation when I integrate those samples per frame. Whereas, the rad/s from the gyro when samples are integrated does translate to a real world rotation.

Chris45215, 2020-07-11:
I propose a test to see if the sensor is somehow mis-calibrated or the IMU component is misaligned. 

The front of the sensor is completely flat, so you can easily stand it up on the front. Try doing so, and the accelerometer should give you a steady value around +9.8. That, at minimum, gives you a good reference and check on the sensor to ensure that it can sense +X correctly. Standing it on its rear should give you -X; and balancing it on its sides should give the Y values. 

Also, keep in mind that those accelerometer values have a much higher frequency than the camera framerate. From https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification the samples are reported at a rate of 208 Hz, so if you are multiplying by the FPS then you will get artificially low values.

neilDGD, 2020-07-13:
Just tried that test and yes, getting ~+9.8/~-9.8 on X, on Y I am getting ~9.67 and ~-9.8 .. any reason why +Y would have that discrepancy and would it matter?

Our process for integrating accelerometer samples is simply every render frame (which is 60hz+) we read as many frames as we can from the IMU (coupled accel + gyro samples), and for each sample:

we integrate the distance travelled per sample (m/(s^2)):

double timeSecs =  (thissample.timeStamp_uS - thissample.timeStamp_uS) / 1000000.0;
double timeSecs_2 = timeSecs * timeSecs;

currentIMUAccelSample.x += thissample.x * timeSecs_2;

As I say, the values from the gyro seem to be translating to real world rotations when integrated, and both accel + gyro samples are coupled @ 208hz (easily consumed per second on a 60hz+ render thread), why am I getting such small values for the accelerometer?

Might also just be missing something obvious? This also does not work for Y and Z.

Thanks,

Neil

closed 2020-08-07 by qm13