microsoft/Azure-Kinect-Sensor-SDK #1277: Nuget not work in WinUI3 C++ project

knat, 2020-07-09:
After adding `Microsoft.Azure.Kinect.BodyTracking` nuget to a [WinUI 3](https://docs.microsoft.com/en-us/windows/apps/winui/winui3/get-started-winui3-for-desktop) desktop C++ project,
```
#include <k4a/k4a.h>
#include <k4abt.h>
```
not work.

Define
```
 <Force-Enable-Microsoft-Azure-Kinect-Sensor>true</Force-Enable-Microsoft-Azure-Kinect-Sensor>
 <Force-Enable-Microsoft-Azure-Kinect-BodyTracking>true</Force-Enable-Microsoft-Azure-Kinect-BodyTracking>
```
in the vcxproj fix the problem, I don't know why.

PS: There's a user voice to add [UWP support](https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/37615828-uwp-support), Maybe WinUI 3 desktop app is an acceptable alternative.

qm13, 2020-07-09:
@knat thanks for providing the info and the suggestion to promote WinUI 3 as an alternative to UWP support. We are investigating the reason the two defines addressed the problem.

qm13, 2020-07-14:
@knat what version of VS are you using?

knat, 2020-07-14:
@qm13 VS 2019 Version 16.7.0 Preview 3.1