microsoft/Azure-Kinect-Sensor-SDK #1278: Body tracking with OpenGL not working on Ubuntu

Umarbi, 2020-07-09:
I am trying to run Kinect DK with an Nvidia GPU Quadro P1000 on ubuntu 18.04. I am running into an issue with the body tracking API where the OpenGL context is not getting created for the body tracking API. I haven't found the right solution yet of what I need to do to resolve the issue. Below is Nvidia-sim output as well as the output when I run k4abt_simple_3d_viewer. Anyone has run into similar issues or has suggestions what I should try next?

Thanks

$ nvidia-smi 
Wed Jul  8 20:44:46 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.100      Driver Version: 440.100      CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  Quadro P1000        Off  | 00000000:01:00.0 Off |                  N/A |
|  0%   46C    P8    N/A /  N/A |     17MiB /  4040MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1289      G   /usr/lib/xorg/Xorg                             9MiB |
|    0      1342      G   /usr/bin/gnome-shell                           4MiB |
+-----------------------------------------------------------------------------+

$ **k4abt_simple_3d_viewer** 

 Basic Navigation:

 Rotate: Rotate the camera by moving the mouse while holding mouse left button
 Pan: Translate the scene by holding Ctrl key and drag the scene with mouse left button
 Zoom in/out: Move closer/farther away from the scene center by scrolling the mouse scroll wheel
 Select Center: Center the scene based on a detected joint by right clicking the joint with mouse

 Key Shortcuts

 ESC: quit
 h: help
 b: body visualization mode
 k: 3d window layout

libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Error: OpenGL 4.4 not supported. Please install latest graphics drivers.
[2020-07-08 20:42:58.463] [error] [t=1995] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (154): depth_engine_start_helper(). Depth engine create and initialize failed with error code: 207.
[2020-07-08 20:42:58.463] [error] [t=1995] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (157): depth_engine_start_helper(). OpenGL 4.4 context creation failed. You could try updating your graphics drivers.
[2020-07-08 20:42:58.463] [error] [t=1995] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (160): deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper()
[2020-07-08 20:42:58.463] [error] [t=1995] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (194): depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread()
[2020-07-08 20:42:58.464] [error] [t=1986] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (552): dewrapper_start(). Depth Engine thread failed to start
[2020-07-08 20:42:58.464] [error] [t=1986] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (451): dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start()
[2020-07-08 20:42:58.464] [error] [t=1986] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[2020-07-08 20:42:58.464] [error] [t=1986] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[2020-07-08 20:42:58.464] [error] [t=1986] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (891): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()
Start K4A cameras failed! 
 - (File: /home/vsts/work/1/s/examples/simple_3d_viewer/main.cpp, Function: main, Line: 127)

qm13, 2020-07-09:
Looking at the error log your graphics driver is not compatible with OpenGL 4.4. Have you updated to the latest graphics driver?

Umarbi, 2020-07-10:
The nvidia-driver-440 was showing up as the recommended version and that's
what I have. Should I be installing a different one?

$ ubuntu-drivers devices

== /sys/devices/pci0000:00/0000:00:01.0/0000:01:00.0 ==

modalias : pci:v000010DEd00001CB1sv000010DEsd000011BCbc03sc00i00

vendor   : NVIDIA Corporation

model    : GP107GL [Quadro P1000]

*driver   : nvidia-driver-440 - distro non-free recommended*

driver   : nvidia-driver-435 - distro non-free

driver   : nvidia-driver-390 - distro non-free

driver   : xserver-xorg-video-nouveau - distro free builtin

== /sys/devices/pci0000:00/0000:00:14.3 ==

modalias : pci:v00008086d0000A370sv00008086sd00000030bc02sc80i00

vendor   : Intel Corporation

manual_install: True

driver   : backport-iwlwifi-dkms - distro free

$ nvidia-smi

Fri Jul 10 08:26:52 2020

+-----------------------------------------------------------------------------+

| NVIDIA-SMI 440.100      *Driver Version: 440.100*      CUDA Version: 10.2
    |

|-------------------------------+----------------------+----------------------+

Umarbi, 2020-07-10:
Here is my Xorg.conf file which was generated using nvidia-xconfig.

$ cat /etc/X11/xorg.conf
# nvidia-xconfig: X configuration file generated by nvidia-xconfig
# nvidia-xconfig:  version 440.100

Section "ServerLayout"
    Identifier     "Default Layout"
    Screen         "Default Screen" 0 0
    InputDevice    "Keyboard0" "CoreKeyboard"
    InputDevice    "Mouse0" "CorePointer"
EndSection

Section "InputDevice"
    # generated from default
    Identifier     "Keyboard0"
    Driver         "kbd"
EndSection

Section "InputDevice"
    # generated from default
    Identifier     "Mouse0"
    Driver         "mouse"
    Option         "Protocol" "auto"
    Option         "Device" "/dev/psaux"
    Option         "Emulate3Buttons" "no"
    Option         "ZAxisMapping" "4 5"
EndSection

Section "Device"
    Identifier     "nvidia"
    Driver         "nvidia"
    VendorName     "NVIDIA Corporation"
EndSection

Section "Screen"
    Identifier     "Default Screen"
    Device         "nvidia"
    DefaultDepth    24
    Option         "AllowEmptyInitialConfiguration"
    SubSection     "Display"
        Depth       24
        Modes      "nvidia-auto-select"
    EndSubSection
EndSection

Here is the relevant part of my Xorg log. There is no EE line. 

[     6.450] (II) LoadModule: "nvidia"
[     6.450] (II) Loading /usr/lib/x86_64-linux-gnu/nvidia/xorg/nvidia_drv.so
[     6.453] (II) Module nvidia: vendor="NVIDIA Corporation"
[     6.453]    compiled for 1.6.99.901, module version = 1.0.0
[     6.453]    Module class: X.Org Video Driver
[     6.453] (II) NVIDIA dlloader X Driver  440.100  Fri May 29 08:21:27 UTC 2020
[     6.453] (II) NVIDIA Unified Driver for all Supported NVIDIA GPUs
[     6.454] (II) systemd-logind: releasing fd for 226:1
[     6.454] (II) Loading sub module "fb"
[     6.455] (II) LoadModule: "fb"
[     6.455] (II) Loading /usr/lib/xorg/modules/libfb.so
[     6.455] (II) Module fb: vendor="X.Org Foundation"
[     6.455]    compiled for 1.20.8, module version = 1.0.0
[     6.455]    ABI class: X.Org ANSI C Emulation, version 0.4
[     6.455] (II) Loading sub module "wfb"
[     6.455] (II) LoadModule: "wfb"
[     6.455] (II) Loading /usr/lib/xorg/modules/libwfb.so
[     6.456] (II) Module wfb: vendor="X.Org Foundation"
[     6.456]    compiled for 1.20.8, module version = 1.0.0
[     6.456]    ABI class: X.Org ANSI C Emulation, version 0.4
[     6.456] (II) Loading sub module "ramdac"
[     6.456] (II) LoadModule: "ramdac"
[... 91 lines cut ...]
[     7.035] (II) Initializing extension RENDER
[     7.035] (II) Initializing extension RANDR
[     7.035] (II) Initializing extension COMPOSITE
[     7.035] (II) Initializing extension DAMAGE
[     7.035] (II) Initializing extension MIT-SCREEN-SAVER
[     7.035] (II) Initializing extension DOUBLE-BUFFER
[     7.035] (II) Initializing extension RECORD
[     7.035] (II) Initializing extension DPMS
[     7.035] (II) Initializing extension Present
[     7.035] (II) Initializing extension DRI3
[     7.035] (II) Initializing extension X-Resource
[     7.035] (II) Initializing extension XVideo
[     7.035] (II) Initializing extension XVideo-MotionCompensation
[     7.035] (II) Initializing extension SELinux
[     7.035] (II) SELinux: Disabled on system
[     7.035] (II) Initializing extension GLX
[     7.035] (II) Initializing extension GLX
[     7.035] (II) Indirect GLX disabled.
[     7.035] (II) GLX: Another vendor is already registered for screen 0
[     7.035] (II) Initializing extension XFree86-VidModeExtension
[     7.036] (II) Initializing extension XFree86-DGA
[     7.036] (II) Initializing extension XFree86-DRI
[     7.036] (II) Initializing extension DRI2
[     7.036] (II) Initializing extension NV-GLX
[     7.036] (II) Initializing extension NV-CONTROL

jmichon, 2020-07-13:
If you're running over ssh, try setting the DISPLAY environment variable. 

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/810

Umarbi, 2020-07-13:
I am doing it over ssh and DISPLAY is set and x forwarding etc. is working. Thanks

wes-b, 2020-07-13:
What Version of OpenGL is reported when you run ```glxinfo | grep "OpenGL"```. Our error message is saying you need version 4.4.

Umarbi, 2020-07-13:
It doesn't even go that far. 
$ glxinfo 
name of display: localhost:10.0
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  149 (GLX)
  Minor opcode of failed request:  6 (X_GLXIsDirect)
  Serial number of failed request:  31
  Current serial number in output stream:  30

Do these look incorrect? I am not sure which numbers I should be seeing here.
$ dpkg -l | grep ^ii | grep -i glx
ii  libgl1-mesa-dev:amd64                      19.2.8-0ubuntu0~18.04.3                          amd64        free implementation of the OpenGL API -- GLX development files
ii  libgl1-mesa-glx:amd64                      19.2.8-0ubuntu0~18.04.3                          amd64        transitional dummy package
ii  libglx-mesa0:amd64                         19.2.8-0ubuntu0~18.04.3                          amd64        free implementation of the OpenGL API -- GLX vendor library
ii  libglx-mesa0:i386                          19.2.8-0ubuntu0~18.04.3                          i386         free implementation of the OpenGL API -- GLX vendor library
ii  libglx0:amd64                              1.0.0-2ubuntu2.3                                 amd64        Vendor neutral GL dispatch library -- GLX support
ii  libglx0:i386                               1.0.0-2ubuntu2.3                                 i386         Vendor neutral GL dispatch library -- GLX support
ii  libnvidia-cfg1-440:amd64                   440.100-0ubuntu0.18.04.1                         amd64        NVIDIA binary OpenGL/GLX configuration library
ii  libnvidia-gl-440:amd64                     440.100-0ubuntu0.18.04.1                         amd64        NVIDIA OpenGL/GLX/EGL/GLES GLVND libraries and Vulkan ICD
ii  libnvidia-gl-440:i386                      440.100-0ubuntu0.18.04.1                         i386         NVIDIA OpenGL/GLX/EGL/GLES GLVND libraries and Vulkan ICD
ii  libxcb-glx0:amd64                          1.13-2~ubuntu18.04                               amd64        X C Binding, glx extension
ii  libxcb-glx0:i386                           1.13-2~ubuntu18.04                               i386         X C Binding, glx extension
ii  libxcb-glx0-dev:amd64                      1.13-2~ubuntu18.04                               amd64        X C Binding, glx extension, development files

I had installed them like this
sudo apt-get install cmake libx11-dev xorg-dev libglu1-mesa-dev freeglut3-dev libglew1.5 libglew1.5-dev libglu1-mesa libglu1-mesa-dev libgl1-mesa-glx libgl1-mesa-dev

Umarbi, 2020-07-13:
Thanks everyone. I just tried it on the attached monitor to the machine and it is working now. Not sure which change made it work. Must have been an ssh env related issue. btw below was the output from glxinfo.

$ glxinfo | grep "OpenGL"
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: Quadro P1000/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 440.100
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6.0 NVIDIA 440.100
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 440.100
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

closed 2020-07-13 by Umarbi

rajkundu, 2024-05-28:
I also had an Azure Kinect SDK Error 207/failure to load `swrast`.

It turned out to be an issue with Anaconda, and [this](https://github.com/spyder-ide/spyder/issues/21450#issuecomment-1781406180) GitHub comment helped immensely. It recommended using conda-forge packages when creating the conda environment (i.e., `conda create --name my_env -c conda-forge python=3.11`). It worked for me!