microsoft/Azure-Kinect-Sensor-SDK #1279: document the Kinect's MJPEG chroma subsampling (either 4:4:4, 4:2:2, or 4:2:0)

diablodale, 2020-07-09:
I request the MJPEG [chroma subsampling](https://en.wikipedia.org/wiki/Chroma_subsampling) for `K4A_IMAGE_FORMAT_COLOR_MJPG` be documented for both clarity and for coding needs.

* Clarity adds another sentence to https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification to disclose the Kinect Azure's sampling.
* Coding allows devs to use specific accelerated decoding methods to process the data and to write specific test cases for CI pipelines. Without knowledge of the subsample, we have to include code for all three, test all three, etc.

My casual inspection of MJPEG frames from `SDK v1.4.0` and Kinect rgb camera firmware `1.6.108` indicates 4:2:2 subsampling.

qm13, 2020-07-14:
Correct. The MJPEG encode supports YCbCr 4:2:2.

qm13, 2020-12-16:
Updated documentation.

closed 2020-12-16 by qm13

diablodale, 2020-12-16:
Hi. I don't see the doc change at https://github.com/MicrosoftDocs/azure-docs/blob/master/articles/kinect-dk/hardware-specification.md

Has the change not been pushed? My intention is to verify the closure. :-)