microsoft/Azure-Kinect-Sensor-SDK #1280: How to check capture contains depth image?

knat, 2020-07-10:
[k4abt_tracker_enqueue_capture()](https://microsoft.github.io/Azure-Kinect-Body-Tracking/release/1.x.x/group__btfunctions_ga093becd9bb4a63f5f4d56f58097a7b1e.html#ga093becd9bb4a63f5f4d56f58097a7b1e) will fail if the `sensor_capture_handle` parameter doesn't contain depth image. So I want to check it:
```C
//capture contains color and/or depth image
k4a_image_t depthImage = k4a_capture_get_depth_image(capture);
if (depthImage != nullptr) {
    auto res = k4abt_tracker_enqueue_capture(tracker, capture, 0);
    k4a_image_release(depthImage);
}
```
My question is: Is it efficient? In my guess, `k4abt_tracker_enqueue_capture()`  will call `k4a_capture_get_depth_image()` too, so there's two `k4a_capture_get_depth_image()` calls.

qm13, 2020-07-13:
This is the most efficient approach with the Body Tracking SDK. Alternatively you could use ```synchronized_images_only``` to guarantee every capture has a depth image.

knat, 2020-07-13:
@qm13 thanks. I'd like to see `bool k4a_capture_has_color_image(k4a_capture_t  capture)` `bool k4a_capture_has_depth_image(k4a_capture_t  capture)` `bool k4a_capture_has_ir_image(k4a_capture_t  capture)` APIs to be added.

closed 2020-07-13 by knat