microsoft/Azure-Kinect-Sensor-SDK #1281: k4a_image_get_gain_speed needs to be added as an API

wes-b, 2020-07-10:
**Is your feature request related to a problem? Please describe.**
Gain can not be read on a frame by frame basis. So there is no way to know when dynamic gain changes are reflected in the streaming images.

**Describe the solution you'd like**
Add k4a_image_get_gain_speed and k4a_image_set_gain_speed

**Describe alternatives you've considered**
API to set and get gain on the sensor doesn't report when gain changes are applied

**Additional context**
N/A

wes-b, 2020-07-10:
gain and ISO speed a tied together. So as work around the user should see iso speed change as gain is applied.

Suruchi-cmd, 2023-02-06:
> gain and ISO speed a tied together. So as work around the user should see iso speed change as gain is applied.

Is there any mathematical relationship that can support this statement or concept?