microsoft/Azure-Kinect-Sensor-SDK #1282: Get intrinsic  from Azure Kinect?

www158, 2020-07-13:
Hello,
I've got a depth image , how to convert it to a point cloud. I can get the intrinsic of the color camera, but i don't know the intrinsic of the depth camera.  Can anyone who has had success with this share any tips, or share the intrinsic for Kinect? Any parameters I can adjust in my case to get it running?

www158, 2020-07-13:
There is another question, can you get a color image and a depth image with the same resolution?

qm13, 2020-07-13:
The calibration blob includes all intrinsic and extrinsic calibration info for color camera, depth camera and IMU. The SDK also includes an API to generate a point cloud. Consider sharing sample code showing what you are attempting to so we can better understand your problem.

www158, 2020-07-17:
@qm13  Hello,Thank you for your reply.
About getting the intrinsic of the depth camera.I run:
```
k4a::calibration calibration = device.get_calibration(config.depth_mode, config.color_resolution);
k4a_calibration_intrinsic_parameters_t *intrinsics = &calibration.depth_camera_calibration.intrinsics.parameters;
```
I set `config.depth_mode = K4A_DEPTH_MODE_WFOV_UNBINNED`，But I can't get the intrinsic through the above code.(other intrinsic can be obtained).Following are the codes and errors,Thanks for your answers:

k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
config.camera_fps = K4A_FRAMES_PER_SECOND_30;
config.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;
config.color_resolution = K4A_COLOR_RESOLUTION_1440P;
config.depth_mode = K4A_DEPTH_MODE_WFOV_UNBINNED;
config.synchronized_images_only = true;
        
k4a::calibration calibration = device.get_calibration(config.depth_mode, config.color_resolution);
k4a_calibration_intrinsic_parameters_t *intrinsics = &calibration.depth_camera_calibration.intrinsics.parameters;

//get the intrinsic
	Eigen::Matrix3f intrinsics_eigen;
	intrinsics_eigen <<
		intrinsics->param.fx, 0.0f, intrinsics->param.cx,
		0.0f, intrinsics->param.fy, intrinsics->param.cy,
		0.0f, 0.0f, 1.0f;
	cout << intrinsics_eigen << std::endl;

[image]

www158, 2020-07-17:
@qm13 Hello,Thank you for your reply.
About getting the intrinsic of the depth camera.I run:

```
k4a::calibration calibration = device.get_calibration(config.depth_mode, config.color_resolution);
k4a_calibration_intrinsic_parameters_t *intrinsics = &calibration.depth_camera_calibration.intrinsics.parameters;
```
I set `config.depth_mode = K4A_DEPTH_MODE_WFOV_UNBINNED`，But I can't get the intrinsic through the above code.(other intrinsic can be obtained).Following are the codes and errors,Thanks for your answers:

```
k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
config.camera_fps = K4A_FRAMES_PER_SECOND_30;
config.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;
config.color_resolution = K4A_COLOR_RESOLUTION_1440P;
config.depth_mode = K4A_DEPTH_MODE_WFOV_UNBINNED;
config.synchronized_images_only = true;

k4a::calibration calibration = device.get_calibration(config.depth_mode, config.color_resolution);
k4a_calibration_intrinsic_parameters_t *intrinsics = &calibration.depth_camera_calibration.intrinsics.parameters;

//get the intrinsic
Eigen::Matrix3f intrinsics_eigen;
intrinsics_eigen <<
intrinsics->param.fx, 0.0f, intrinsics->param.cx,
0.0f, intrinsics->param.fy, intrinsics->param.cy,
0.0f, 0.0f, 1.0f;
cout << intrinsics_eigen << std::endl;
```
[image]

closed 2020-08-07 by qm13

UEBoy2019, 2021-01-23:
@qm13  it will have different intrinsic if depth _mode is not same