microsoft/Azure-Kinect-Sensor-SDK #1283: Sensor not detected upon first boot on Jetson Nano

jmichon, 2020-07-13:
**Describe the bug**

The device fails to run on the first try on the Jetson Nano. `k4aviewer` fails to detect the device.  
Kinect is connected to power using the attached DC adapter. Factory cables are used.  
After resetting Kinect's power or rebooting Jetson, everything works correctly. 

**To Reproduce**

1. Connect Kinect to the Jetson Nano (both have to be unplugged from power).
2. Connect Kinect to power.  
3. Turn on Jetson Nano. 
4. Run k4aviewer (error appears). 
5. Unplug Kinect's power, plug in again. 
6. Run k4aviewer.  

**Expected behavior**

Device runs correctly every time. 

**Logs**
```[2020-07-13 14:22:42.452] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (53): k4a_device_t_create(). Created   k4a_device_t 0x55946c33d0
[2020-07-13 14:22:42.453] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (29): depthmcu_t_create(). Created   depthmcu_t 0x55946d99d0
[2020-07-13 14:22:42.453] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usb_cmd_priv.h (101): usbcmd_t_create(). Created   usbcmd_t 0x55946d87e0
[2020-07-13 14:22:42.465] [info] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (343): find_libusb_device(). Container ID found: {cbdf6740-ba76-442e-8086-d58f4dbf9602} 
[2020-07-13 14:22:42.465] [info] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (224): populate_serialnumber(). Serial Number found 000875194512
[2020-07-13 14:22:42.465] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (653): usb_cmd_io(). XFR: Cmd=00000115, PayloadSize=255
[2020-07-13 14:22:42.480] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color_mcu/color_mcu.c (41): colormcu_t_create(). Created   colormcu_t 0x55946dbb40
[2020-07-13 14:22:42.480] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usb_cmd_priv.h (101): usbcmd_t_create(). Created   usbcmd_t 0x55946e17a0
[2020-07-13 14:22:42.559] [info] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (343): find_libusb_device(). Container ID found: {cbdf6740-ba76-442e-8086-d58f4dbf9602} 
[2020-07-13 14:22:42.559] [info] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (224): populate_serialnumber(). Serial Number found 000875194512
[2020-07-13 14:22:42.559] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/calibration/calibration.c (47): calibration_t_create(). Created   calibration_t 0x55946dd660
[2020-07-13 14:22:42.559] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (653): usb_cmd_io(). XFR: Cmd=00000111, PayloadSize=10240
[2020-07-13 14:22:44.560] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (702): usb_cmd_io(). Error calling libusb_bulk_transfer for rx, result:LIBUSB_ERROR_TIMEOUT
[2020-07-13 14:22:44.560] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2020-07-13 14:22:44.560] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (412): usb_cmd_read( depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_CALIBRATION_DATA, NULL, 0, (uint8_t *)json, json_size, bytes_read) returned failure in depthmcu_get_extrinsic_calibration()
[2020-07-13 14:22:44.560] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (653): usb_cmd_io(). XFR: Cmd=00000111, PayloadSize=15360
[2020-07-13 14:22:46.561] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (748): usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_TIMEOUT
[2020-07-13 14:22:46.562] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2020-07-13 14:22:46.563] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (412): usb_cmd_read( depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_CALIBRATION_DATA, NULL, 0, (uint8_t *)json, json_size, bytes_read) returned failure in depthmcu_get_extrinsic_calibration()
[2020-07-13 14:22:46.563] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (653): usb_cmd_io(). XFR: Cmd=00000111, PayloadSize=20480
[2020-07-13 14:22:48.564] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (748): usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_TIMEOUT
[2020-07-13 14:22:48.564] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2020-07-13 14:22:48.564] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (412): usb_cmd_read( depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_CALIBRATION_DATA, NULL, 0, (uint8_t *)json, json_size, bytes_read) returned failure in depthmcu_get_extrinsic_calibration()
[2020-07-13 14:22:48.564] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (653): usb_cmd_io(). XFR: Cmd=00000111, PayloadSize=25600
[2020-07-13 14:22:50.564] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (748): usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_TIMEOUT
[2020-07-13 14:22:50.564] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2020-07-13 14:22:50.564] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (412): usb_cmd_read( depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_CALIBRATION_DATA, NULL, 0, (uint8_t *)json, json_size, bytes_read) returned failure in depthmcu_get_extrinsic_calibration()
[2020-07-13 14:22:50.565] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (653): usb_cmd_io(). XFR: Cmd=00000111, PayloadSize=30720
[2020-07-13 14:22:52.565] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (748): usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_TIMEOUT
[2020-07-13 14:22:52.565] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2020-07-13 14:22:52.565] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (412): usb_cmd_read( depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_CALIBRATION_DATA, NULL, 0, (uint8_t *)json, json_size, bytes_read) returned failure in depthmcu_get_extrinsic_calibration()
[2020-07-13 14:22:52.565] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (653): usb_cmd_io(). XFR: Cmd=00000111, PayloadSize=35840
[2020-07-13 14:22:54.565] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (748): usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_TIMEOUT
[2020-07-13 14:22:54.566] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2020-07-13 14:22:54.566] [error] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (412): usb_cmd_read( depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_CALIBRATION_DATA, NULL, 0, (uint8_t *)json, json_size, bytes_read) returned failure in depthmcu_get_extrinsic_calibration()
[2020-07-13 14:22:54.566] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (653): usb_cmd_io(). XFR: Cmd=00000111, PayloadSize=40960
[... cut ...]
[2020-07-13 14:23:02.577] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color_mcu/color_mcu.c (41): colormcu_t_destroy(). Destroyed colormcu_t 0x55946dbb40
[2020-07-13 14:23:02.578] [trace] [t=7197] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (53): k4a_device_t_destroy(). Destroyed k4a_device_t 0x55946c33d0
```

**Desktop (please complete the following information):**
 - OS with Version: NVida JetPack 4.4-b144 / Ubuntu 18.04.4 LTS
 - SDK Version: 1.4.1
 - Firmware version:  
RGB Sensor Version:  1.6.108
Depth Sensor Version:1.6.79
Mic Array Version:   1.6.14
Sensor Config:       6109.7

jmichon, 2020-07-14:
Update: this happens only when Jetson starts with the Kinect attached (and plugged into power). I'm attaching `dmesg` output. 

[dmesg.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4920294/dmesg.log)

jmichon, 2020-07-21:
Another update, dmesg with messages from libusb concerning only K4A. Please confirm if this is an issue with the firmware, with the particular Kinect or the problem lays somewhere in the host. 
```
[    6.223166] usb 2-1.1.1: new SuperSpeed USB device number 4 using tegra-xusb
[    6.243593] usb 2-1.1.1: Isoc endpoint with wBytesPerInterval of 1024 in config 1 interface 1 altsetting 4 ep 129: setting to 944
[    6.257767] usb 2-1.1.1: New USB device found, idVendor=045e, idProduct=097d
[    6.257771] usb 2-1.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    6.257775] usb 2-1.1.1: Product: Azure Kinect 4K Camera
[    6.257778] usb 2-1.1.1: Manufacturer: Microsoft
[    6.257781] usb 2-1.1.1: SerialNumber: 000875194512
[   11.147437] usb 2-1.1.1: usb_suspend_both: status 0
[   29.138843] usb 2-1.1.1: usb_suspend_both: status 0
---------- RUNNING K4ARECORDER - FAILED ----------------------------
[   96.138868] usb 2-1.1.1: usb_suspend_both: status 0
---------- KINECT DISCONNECTED FROM POWER AND RECONNECTED ----------
[  143.644097] usb 2-1.1.1: USB disconnect, device number 4
[  144.496138] usb 2-1.1.1: new SuperSpeed USB device number 7 using tegra-xusb
[  144.519070] usb 2-1.1.1: Isoc endpoint with wBytesPerInterval of 1024 in config 1 interface 1 altsetting 4 ep 129: setting to 944
[  144.533063] usb 2-1.1.1: New USB device found, idVendor=045e, idProduct=097d
[  144.533095] usb 2-1.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  144.533119] usb 2-1.1.1: Product: Azure Kinect 4K Camera
[  144.533141] usb 2-1.1.1: Manufacturer: Microsoft
[  144.533162] usb 2-1.1.1: SerialNumber: 000875194512
[  152.011357] usb 2-1.1.1: usb_suspend_both: status 0
[  153.604426] usb 2-1.1.1: USB disconnect, device number 7
[  155.032702] usb 2-1.1.1: new SuperSpeed USB device number 9 using tegra-xusb
[  155.055920] usb 2-1.1.1: Isoc endpoint with wBytesPerInterval of 1024 in config 1 interface 1 altsetting 4 ep 129: setting to 944
[  155.070036] usb 2-1.1.1: New USB device found, idVendor=045e, idProduct=097d
[  155.070068] usb 2-1.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  155.070092] usb 2-1.1.1: Product: Azure Kinect 4K Camera
[  155.070114] usb 2-1.1.1: Manufacturer: Microsoft
[  155.070135] usb 2-1.1.1: SerialNumber: 000875194512
[  162.987087] usb 2-1.1.1: usb_suspend_both: status 0
---------- RUNNING K4ARECORDER - SUCCESS ---------------------------
[  208.660571] usb 2-1.1.1: usbfs: process 7677 (k4arecorder) did not claim interface 0 before use
[  211.083314] usb 2-1.1.1: usb_suspend_both: status 0
```

jmichon, 2020-08-03:
@qm13 @hardyzhuang any news on this?

asergaz, 2020-08-03:
Related issue: https://docs.microsoft.com/en-us/answers/questions/55354/index.html

metervara, 2020-08-24:
Regarding the questions asked here:
> Related issue: https://docs.microsoft.com/en-us/answers/questions/55354/index.html

> 1. What machine model is the host?

Vendor: ASUSTeK COMPUTER INC.
Version: 0401
Manufacturer: ASUSTeK COMPUTER INC.  
Model: GR8 II  
SystemType: x64-based PC

> 2. Which OS build is running on host?

OS Name: Microsoft Windows 10 Pro
OS Version: 10.0.18362 N/A Build 18362

> 3. Which usb port of host does AKinect connect to?

I think it's Port 1 & 2, but I'm not sure that's exactly the information you're looking for?

I haven't tested the restart + reset camera workaround, I'm on a different PC now and that one does not have this issue. 

Sorry for the late reply, I've been on vacation.

sjdrc, 2020-09-25:
We've been experiencing this exact issue with the LattePanda Delta. It's causing us significant problems. Can someone from the kinect team please give an update on this?

sjdrc, 2020-11-17:
It’s been months and there is no fix or any update from the Kinect team. This issue is significant and affects both the SBC we currently use in production (LattePanda Delta), and our potential replacement for it (Jetson nano). We need urgent answers to these questions:
- What is the cause of this issue, is it with the Jetsons or with the K4A DKs?
- Is there a solution actively been worked on and when can we expect that?
- Is there a workaround we can implement in the meantime to temporary mitigate this issue?

sjdrc, 2020-12-02:
@qm13

hardyzhuang, 2020-12-07:
It took us time to identify the root cause. Fortunately, we have had a fix in regression test now. It is a USB stack low-level fix, will take some time. Please stay tuned...thanks,

jmichon, 2021-02-12:
@hardyzhuang @qm13 Any updates on this?

ben-kraus, 2021-02-16:
Can reproduce the issue with Jetson Xavier NX and Jetpack 45.

shariq, 2021-03-29:
Bump on this; it's a pretty serious issue for us :(

blakerutledge, 2021-04-05:
Bump for us as well, this one is kinda mission critical

maggieroxas, 2021-05-19:
Bump, too. The USB/power connections (and Nano) are inside our robot, it is unusual that we have to open up the robot every power-up to access these cables and reconnect the Azure for it to work...

jplapp, 2021-06-30:
we ran into the same issue. However, running `AzureKinectFirmwareTool -Reset` works for us on the jetson.

jmichon, 2021-07-01:
@qm13 @hardyzhuang It's been over 6 months now since you've reported about the fix being in testing, any chance of releasing it?

qm13, 2021-07-19:
Please test with the firmware beta version 1.6.110080014 located at [firmware-beta](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/firmware-beta)

Aquariusquintini, 2022-02-01:
> Please test with the firmware beta version 1.6.110080014 located at [firmware-beta](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/firmware-beta)

The issue is NOT fixed with the latest firmware version 1.6.110080014. At least not on my x64-based Notebook running Windows 10. Testet with multiple cameras, but the same result. The firmware tool reset seems to work, but is not a permanent solution.

What machine model is the host?
Vendor: Dell Latitude 5421
SystemType: x64-based PC

Which OS build is running on host?
OS Name: Microsoft Windows 10 Pro
OS Version: 21H1 Build 19043.1466

Which usb port of host does AKinect connect to?
The problem occurs independent of the used USB-Port. Especially after reconnecting the USB-Plug or about once every 15 attempts to connect to the device.

SDK Version 1.4.1

sjdrc, 2022-02-01:
We've encountered this issue and the beta firmware resolved the reset issue on jetsons, but we never saw this exhibited on an x86_64 machine.

BryanPlayLu, 2022-03-09:
We have also been facing this same issue on x64 Windows PCs where the K4A device would stop being detected and prints an error from find_libusb_device() when trying to open the kinect using the SDK.. 

It's also not a possibility for our users to physically unplug/plug the usb device everytime this happens, we are trying to find a solution that would work via software.

When this happens, "AzureKinectFirmwareTool -Reset" seems to work everytime for the device to be detected again but we're still trying to figure out if there's a way to do the equivalent via the SDK, without using the firmware tool

SeanTasker, 2023-09-05:
We came up with a work around on Linux that involved resetting the port using a shell command. This was a couple of year ago now but our wiki says we used https://github.com/mvp/uhubctl and ran:

```
sudo uhubctl -l 2-1 -p 1 -a cycle
```

My note on the page includes
> That will reset the USB power might be nearly instant, might be around 4-5 mins

joao-corvo, 2023-09-06:
The fact that the reset can take 4-5 min, is not something that we would want in a robot which is in the operation site of the costumers :/

SeanTasker, 2023-09-06:
It definitely isn't ideal. We never looked into the cause of the delay as we were dealing with a fleet of them mounted stationary. Running `perf top` might tell you if there is a kernel call being polled.

This was a couple of years ago too, so whatever was causing the issue for us (it may have been related to our software) may be resolved in more recent software.

Good luck!

joao-corvo, 2023-09-06:
I just happened this https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1886 to us right now. Any solution regarding this issue till now?