microsoft/Azure-Kinect-Sensor-SDK #1292: C# SDK cannot open 2 devices in the same thread

jregele, 2020-07-22:
I'm trying to get 2 cameras to work in sync using the C# SDK. I was able to get the green screen solution working just fine and there was no separate threads for each camera. Also I can get both cameras working in C#, but they have to be opened and started in separate threads, which will cause headaches reading from both cameras in the same domain.

Basically, when I do this

```
primary = Device.Open(primary_idx);
secondary = Device.Open(secondary_idx);
```

I get the following error.

```
libusb: error [winusbx_open] could not open device \\.\USB#VID_045E&PID_097D&MI_02#8&24C7C21B&0&0002#{875CFCC8-678E-4C12-BA03-5119D706A3F0} (interface 2): [5] Access is denied.
libusb: warning [handle_timeout] async cancel failed -5 errno=0
libusb: warning [handle_timeout] async cancel failed -5 errno=0
libusb: warning [handle_timeout] async cancel failed -5 errno=0
libusb: warning [handle_timeout] async cancel failed -5 errno=0
libusb: warning [handle_timeout] async cancel failed -5 errno=0
libusb: warning [handle_timeout] async cancel failed -5 errno=0
libusb: warning [handle_timeout] async cancel failed -5 errno=0
```

I have tried using Thread.Sleep between the calls in case it was some sort of race condition but it didn't help.

Also opening either Device works on its own, just the second device can never be opened until after calling Dispose() on the first.

qm13, 2020-07-27:
Access denied implies that the camera is seen but cannot be opened. Please verify that you do not have an open instance of the camera elsewhere.

sandris-b, 2020-07-28:
I'm opening 3 devices in same thread and its working fine
` for (int x = 0; x < CameraCount; x++)
            {
                try
                {
                    device = Device.Open(x);`

Maybe your problem is in camera configuration?
here is mine, which is working fine
` public static class KinectConfiguration
    {
        // Two depth images should be seperated by MIN_TIME_BETWEEN_DEPTH_CAMERA_PICTURES_USEC to ensure the depth imaging
        // sensor doesn't interfere with the other. To accomplish this the master depth image captures
        // (MIN_TIME_BETWEEN_DEPTH_CAMERA_PICTURES_USEC / 2) before the color image, and the subordinate camera captures its
        // depth image (MIN_TIME_BETWEEN_DEPTH_CAMERA_PICTURES_USEC / 2) after the color image. This gives us two depth
        // images centered around the color image as closely as possible.

        private const int MIN_TIME_BETWEEN_DEPTH_CAMERA_PICTURES_USEC = 1600;//10 ticks = 1000 nano sec = 1 usec

        private static readonly DeviceConfiguration DEFAULT_CONFIGURAITION = new DeviceConfiguration
        {
            ColorFormat = ImageFormat.ColorBGRA32,
            ColorResolution = ColorResolution.R1080p,
            DepthMode = DepthMode.NFOV_2x2Binned,
            CameraFPS = FPS.FPS30,
            SynchronizedImagesOnly = true,
            SuboridinateDelayOffMaster = TimeSpan.Zero
        };

        public static DeviceConfiguration GetConfiguration(KinectType kinectType)
        {
            var result = DEFAULT_CONFIGURAITION;
            switch (kinectType)
            {
                case KinectType.MasterLeft:
                    result.WiredSyncMode = WiredSyncMode.Master;
                    result.DepthDelayOffColor = TimeSpan.Zero;
                    break;
                case KinectType.Right:
                    result.WiredSyncMode = WiredSyncMode.Subordinate;
                    result.DepthDelayOffColor = TimeSpan.FromTicks(-MIN_TIME_BETWEEN_DEPTH_CAMERA_PICTURES_USEC);
                    break;
                case KinectType.Top:
                    result.WiredSyncMode = WiredSyncMode.Subordinate;
                    result.DepthDelayOffColor = TimeSpan.FromTicks(MIN_TIME_BETWEEN_DEPTH_CAMERA_PICTURES_USEC);
                    break;
            }
            return result;
        }
    }`

Chris45215, 2020-08-03:
I'm able to open 5 cameras in the same thread. But I do not recommend it. The way the cameras are set up - and because of issues in the body tracking SDK (if you are using body tracking) - it will add extra latency (especially if you use the same DNN instance shared among all the cameras). Microsoft's tutorials do not include instructions for multi-camera code, as far as I know.

closed 2020-08-07 by qm13