microsoft/Azure-Kinect-Sensor-SDK #1294: [Firmware / Hardware] Some Kinect devices don't initialise via USB-C -> C

37, 2020-07-23:
Hi, I think I've got a fun one for you. 

I have identified an issue whereby some Azure Kinect devices will not attempt initialise when plugged in to a machine that I have proven to work with other Azure Kinect devices. So far, I have found this issue presents in about 12 of 122 Kinect devices tested. 

The machines I'm using are Lattepanda Delta 432 units running Ubuntu 18.04. Unless I mention otherwise I am connecting and powering Kinects via USB-C on the host machine. 

The issue: 
If I plug a problematic Kinect into one of my machine sometimes no Kinect status light appears (totally blank; same behaviour as when you solely power via the DC-barrel jack without a USB) and no devices appear in lsusb. 

```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 8087:0aaa Intel Corp. 
Bus 001 Device 006: ID 2341:8036 Arduino SA Leonardo (CDC ACM, HID)
Bus 001 Device 005: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 004: ID 04e2:1410 Exar Corp. XR21V1410 USB-UART IC
Bus 001 Device 003: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 002: ID 1a2c:0e24 China Resource Semico Co., Ltd 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
I get the same output with no Kinect plugged in. 

Further, dmesg -w remains completely silent and it's as if no device is connected at all. I'll refer to this Kinect, and ones that demonstrate the same behaviour as 'problem Kinects'. 

Where this gets interesting is that with this exact same machine, I can plug in a different Kinect with the same usb and it will function perfectly. lsusb will show 5 devices, and I can use the sensor. 

Additionally, some machines work with *any* kinect, and 'problem Kinects' will always work with a laptop.  

I have spent today trying to debug this on 9 kinect devices, testing with 4 machines. 2 of the machines selected work with any Kinect, and two don't. *All 4 machines are the same Lattepanda model and have the same Intel usb host controller*. 

## Test machine information: 

Machine name | kernel
-- | --
Machine 1 | Linux blk1-c 5.3.0-51-generic #44~18.04.2-Ubuntu SMP Thu Apr 23 14:27:18 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux |
Machine 2 | Linux blk1-h 5.3.0-51-generic #44~18.04.2-Ubuntu SMP Thu Apr 23 14:27:18 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux |
Machine 3 | Linux hfm1-3 5.3.0-62-generic #56~18.04.1-Ubuntu SMP Wed Jun 24 16:17:03 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux |
Machine 4 | Linux hfm1-4 5.3.0-61-generic #55~18.04.1-Ubuntu SMP Mon Jun 22 16:40:20 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux | 

Note: I formatted and reflashed machine 4 and machine 1 to the latest kernel and saw no changes. 

## Test Kinect information: 

Kinect | RGB firmware | Depth firmware | Audio firmware | Build
-- | -- | -- | -- | --
A | 1.6.102 | 1.6.75 | 1.6.14 | Release
B | 1.6.102 | 1.6.75 | 1.6.14 | Release
C | 1.6.102 | 1.6.75 | 1.6.14 | Release
D | 1.6.102 | 1.6.75 | 1.6.14 | Release
E | 1.6.102 | 1.6.75 | 1.6.14 | Release
F | 1.6.102 | 1.6.75 | 1.6.14 | Release
G | 1.6.102 | 1.6.75 | 1.6.14 | Release
H | 1.6.102 | 1.6.75 | 1.6.14 | Release
I | 1.6.102 | 1.6.75 | 1.6.14 | Releas

## Testing standard usb-c. 

Lattepanda | Kinect A | Kinect B | Kinect C | Kinect D | Kinect E | Kinect F | Kinect G | Kinect H | Kinect I
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
Machine 1 | 5 | 5 | 5 | 5 | 5 | 5 | 5 | 5 | 5
Machine 2 | 5 | 5 | 5 | 5 | 5 | 5 | 5 | 5 | 5
Machine 3 | 5 | 0 | 0 | 0 | 0 | 0 | 5 | 5 | 5
Machine 4 | 5 | 0 | 0 | 0 | 0 | 0 | 5 | 5 | 5

Above: cell numbers are the count of Microsoft devices that appear in the output of `lsusb | grep Microsoft`. 5 is good :+1:. These numbers can be replicated and remain the same even if the usb is left in, machine rebooted, booted first then plugged in, etc. I've tried also endless USB combinations. 

## Testing  Direct DC power + USBC data 

Lattepanda | Kinect A | Kinect B | Kinect C | Kinect D | Kinect E | Kinect F | Kinect G | Kinect H | Kinect I
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
Machine 1 | solid white | solid white | solid white | solid white | solid white | solid white | solid white | solid white | solid white
Machine 2 | solid white | solid white | solid white | solid white | solid white | solid white | solid white | solid white | solid white
Machine 3 | solid white | None | None | None | None | None | solid white | solid white | solid white
Machine 4 | solid white | None | None | None | None | None | solid white | solid white | solid white

Above: status light colours on the connected Kinect device. 

## Testing no DC power + USBA data cable
 
Lattepanda | Kinect A | Kinect B | Kinect C | Kinect D | Kinect E | Kinect F | Kinect G | Kinect H | Kinect I
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
Machine 1 | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber
Machine 2 | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber
Machine 3 | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber
Machine 4 | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber | Flasing amber

## Testing both power & over USB-A (DB barrel jack cable + A->C) 

Lattepanda | Kinect A | Kinect B | Kinect C | Kinect D | Kinect E | Kinect F | Kinect G | Kinect H | Kinect I
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
Machine 1 | 5 | 5 | 5 | 5 | 5 | 5 | 5 | 5 | 5
Machine 2 | 5 | 5 | 5 | 5 | 5 | 5 | 5 | 5 | 5
Machine 3 | 5 | 5 | 5 | 5 | 5 | 5 | 5 | 5 | 5
Machine 4 | 5 | 5 | 5 | 5 | 5 | 5 | 5 | 5 | 5

Above: All Kinect devices work with all machines when powered and connected over two dedicated USB-A cables. This is a fine workaround for testing though unfortunately I need to use the machine's USB-C to run the Kinect for my application. I also need to use these Lattepanda boards.

## Testing updating Kinect firmware 

I updated `Kinect B` a known 'problem' unit to the latest firmware (`AzureKinectDK_Fw_1.6.110079014` from ) 

# Still writing this ticket. 
Submitting so I don't lose progress on tab crash.

I have no idea where the root cause resides, though my gut says there's some trickery around voltage tolerances within the Lattepanda usb host controller or within the Kinect's internal USB 3.0 hub.

37, 2020-07-30:
Hey @qm13 if you want me to post you some of the hardware that I've found issues with let me know. Happy to do so if it'll help in debugging :surfer:

qm13, 2020-08-29:
Handling offline

closed 2020-08-29 by qm13