microsoft/Azure-Kinect-Sensor-SDK #1295: how to update opengl version on ubuntu

Apollo19971206, 2020-07-27:
When I run k4aviewer on Ubuntu18.04, the error occurd due to my OpenGL version is too low just like#918 .

I install nvidia driver by download run file frome NVIDIA website which requires not to install opengl files.
 `sudo sh ./xxx.run --no-opengl-files` 

So how can I update my opengl version?

qm13, 2020-07-27:
Our experience is that OpenGL is installed with the NVIDIA driver. Recommend you upgrade the driver and let it install OpenGL.

Apollo19971206, 2020-07-28:
> Our experience is that OpenGL is installed with the NVIDIA driver. Recommend you upgrade the driver and let it install OpenGL.

OK. Thanks for your quickly reply.

drewm1980, 2020-08-05:
We're off topic, but I also had to update my nvidia driver to get a new enough openGL version. I recommend dong it through apt, not with nvidia's runfiles:

* first add nvidia repository
sudo add-apt-repository ppa:graphics-drivers/ppa
sudo apt update
* list possible drivers
sudo ubuntu-drivers devices
* do a upgrade
sudo apt upgrade
* install recommended driver (this was the 415 on my machine)
sudo ubuntu-drivers autoinstall
* reboot

closed 2020-08-07 by qm13