microsoft/Azure-Kinect-Sensor-SDK #1296: undefined reference to `k4a_playback.... error when compile body-traking-samples using CMake

haeginh, 2020-07-27:
**Describe the bug**
Even after installing all the prerequisites described in readme and the Azure Kinect website, error occured when I complied Azure-Kinect-Samples by exactly follwing the instruction.
The error message is as below
```
CMakeFiles/simple_3d_viewer.dir/main.cpp.o: In function `PlayFile(InputSettings)':
main.cpp:(.text+0xd72): undefined reference to `k4a_playback_open'
main.cpp:(.text+0xdb9): undefined reference to `k4a_playback_get_calibration'
main.cpp:(.text+0xf2d): undefined reference to `k4a_playback_get_next_capture'
main.cpp:(.text+0x1110): undefined reference to `k4a_playback_close'
collect2: error: ld returned 1 exit status
```

**To Reproduce**
Simply try to build Azure-Kinect-Samples in Ubuntu 18.04 just as decribed in readme

**Expected behavior**
The error occurs when it tries to link executable simple_3d_viewer, and dose not create the executable.

**Additional context**
This could be resolved by adding k4arecord in target_link_libraries in CMakeLists.txt of simple_3d_viewer. Also it would be nice if it is possible to compile the whole examples by adding all the subdirectories in CMakeLists.txt of body-tracking-samples.
+ For Ubuntu repository, there is some inconsistency; k4a-tools is installed with libk4a1.4, while libk4abt1.0 is installed with libk4a1.3. So... it is currently impossible to apt install k4a-tools and libk4abt at the same time. I didn't make a separate issue for this since I found similar previous one.

AswinkarthikeyenAK, 2020-11-10:
Hi @haeginh ,
Were you able to launch simple_3d_viewer? I tried your workaround, and it helped in building the executable, but I cannot launch the simple_3d_viewer.
Can you please let me know it worked for you?

Thanks

dwhite54, 2021-03-17:
For the Ubuntu repo issue, I managed to work around it as follows (warning: **ugly**):

1. `sudo apt --purge remove k4a-tools libk4abt1.0-dev libk4a1.3 libk4a1.4`
2. `sudo apt install k4a-tools`
3. `apt download libk4abt1.0-dev`
4. `apt download libk4abt1.0`
6. `sudo dpkg --ignore-depends=libk4a1.3,libk4a1.3-dev -i libk4abt1.0_1.0.0_amd64.deb`
7. `sudo dpkg --ignore-depends=libk4a1.3,libk4a1.3-dev -i libk4abt1.0-dev_1.0.0_amd64.deb`
8. `sudo ln -s /usr/lib/x86_64-linux-gnu/libk4a.so.1.4 /usr/lib/x86_64-linux-gnu/libk4a.so.1.3`

Even still, apt will complain about the unmet dependencies. My (**more ugly**) solution for that is:

1. Edit `/var/lib/dpkg/status`, 
2. find `Package: libk4a1.3` and `Package: libk4a1.3-dev`
3. Replace those blocks with 
```
Package: libk4a1.3
Status: install ok installed
Priority: optional
Section: devel
Maintainer: Microsoft
Architecture: amd64
Version: 1.4.1
Description: Fake package listing for "Dynamic Libraries for Azure Kinect Runtime"

Package: libk4a1.3-dev
Status: install ok installed
Priority: optional
Section: devel
Maintainer: Microsoft
Architecture: amd64
Version: 1.4.1
Description: Fake package listing for "Headers and cmake files needed for Azure Kinect Development"
```
4. Also, for `Package: libk4abt1.0` change the `Depends` lines to specify `libk4a1.4 (>= 1.3.0)`
5. and `Package: libk4abt1.0-dev` ... `Pre-Depends: libk4a1.4 (>= 1.3.0)`
6. Finally, prevent apt from upgrading the body tracking headers with `sudo apt mark-hold libk4abt1.0-dev`

After all this, I am able to run `k4abt_simple_3d_viewer` and apt isn't broken nor itching to "fix" the hacked dependencies.

GaneshBaronAloir, 2021-06-18:
Hello,

I have the same issue and can't get past it. Anyone found a solution?

haeginh, 2021-06-24:
As I mentioned in the additional context, this could be resolved by linking target "k4arecord". Plus, I think now everything works pretty well in new versions :-)

haeginh, 2021-06-24:
> Hi @haeginh ,
> Were you able to launch simple_3d_viewer? I tried your workaround, and it helped in building the executable, but I cannot launch the simple_3d_viewer.
> Can you please let me know it worked for you?
> 
> Thanks

Sorry for the too late reply. Yes, it worked for me.

closed 2021-08-16 by qm13