microsoft/Azure-Kinect-Sensor-SDK #1298: C++ vs C# Camera Extrinsics

Vangos, 2020-07-29:
**Describe the bug**

In C++, I can transform from a source to a target coordinate system by using the 2D `extrinsics` array:

```
calibration.extrinsics[K4A_CALIBRATION_TYPE_ACCEL][K4A_CALIBRATION_TYPE_DEPTH]
```

In C#, the `Extrinsics` array is 1D, thus it's unclear how to transform from one system to another.

**To Reproduce**

1. Create an Azure Kinect C# project or use an existing one.
1. Find your `Device` object.
1. Call `device.GetCalibration().DeviceExtrinsics`.
1. Notice that the array is 1D instead of 2D.

Also, compare the [C++ documentation](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/release/1.4.x/structk4a__calibration__t_a03f4b1600f8088739ffeea818a5df890.html#a03f4b1600f8088739ffeea818a5df890) to the [C# documentation](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/release/1.4.x/struct_microsoft_1_1_azure_1_1_kinect_1_1_sensor_1_1_calibration_a252403b59e173beb558d541e10c6aa61.html).

Both pages state:

> The extrinsic parameters allow 3D coordinate conversions between depth camera, color camera, the IMU's gyroscope and accelerometer. To transform from a source to a target 3D coordinate system, use the parameters stored under extrinsics[source][target].

Am I probably missing something here?

aloyspx, 2021-01-09:
Is there a workaround for this bug in the meantime?

Zumra, 2021-03-15:
Exploring the content of the `device.GetCalibration().DeviceExtrinsics`, it seems to me that the 1D array can be indexed as:
`DeviceExtrinsics[(int)CalibrationDeviceType.Num * (int)source + (int)target]`
where `source` and `target` are of type `CalibrationDeviceType`.
This will obtain the parameters needed to transform from a source to a target 3D coordinate system.

Until there is a fix available, can the AK team confirm this is correct?

Vangos, 2021-03-22:
For anyone reading this, @Zumra's comment is correct. Until there's an official API, you can use the following extension method to transform between a source and a target calibration device:

```C#
public static class CalibrationExtensions
{
    public static Extrinsics Extrinsics(this Calibration calibration, 
        CalibrationDeviceType from,  CalibrationDeviceType to)
    {
        int index = (int)CalibrationDeviceType.Num * (int)from + (int)to;
        return calibration.DeviceExtrinsics[index];
    }
}
```

Usage:

```C#
Device _device = Device.Open();
// ...

Calibration calibration = _device.GetCalibration();
Extrinsics accelToDepth = calibration.Extrinsics(CalibrationDeviceType.Accel, CalibrationDeviceType.Depth);

float[] rotation = accelToDepth.Rotation;
```

That's how you can implement floor tracking in C# 😉