microsoft/Azure-Kinect-Sensor-SDK #1302: Azure Kinect Persistent Crash - Help appreciated

RobbyGraham, 2020-07-30:
I've just migrated to windows specifically for this tech so excuse my ignorance, am not a coder but have been at this 2 days and have reached a real impasse. 

Running Azure Kinect on Windows 10 on Razer Blade Stealth 13 (Specs below) and getting a repeated problem that causes the Kinect to fail ( with standard windows message following - the last USB device you connected to windows has malfunctioned AND WINDOWS DOES NOT RECOGNISE IT). This happens in Kinect Viewer and also when I try to open in DepthKit. Both have different faults (Logs below) but always involve LIBUSB_ERROR or references the DepthSensor (moreso in depthkit)

KINECT VIEWER FAULT

[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 42ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 41ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 37ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 40ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 42ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 39ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 42ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 42ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 36ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 37ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 40ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 35ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 35ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 34ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 36ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 41ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 39ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 34ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 35ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 35ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 35ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 37ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 42ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 37ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 41ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 34ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 34ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 41ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 35ms (this may be transient).
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 39ms (this may be transient).
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000002 unexpected
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000002 unexpected
[ error    ] : usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_IO
[ error    ] : usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, cmd_status) returned failure in usb_cmd_write_with_status()
[ error    ] : usb_cmd_write_with_status(depthmcu->usb_cmd, DEV_CMD_DEPTH_STREAM_STOP, NULL, 0, NULL, 0, &cmd_status) returned failure in depthmcu_depth_stop_streaming()
[ error    ] : usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_IO
[ error    ] : usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, cmd_status) returned failure in usb_cmd_write_with_status()
[ error    ] : usb_cmd_write_with_status(depthmcu->usb_cmd, DEV_CMD_DEPTH_STOP, NULL, 0, NULL, 0, &cmd_status) returned failure in depthmcu_depth_stop_streaming()
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000002 unexpected
[ error    ] : usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_IO
[ error    ] : usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, &cmd_status) returned failure in usb_cmd_write()
[ error    ] : usb_cmd_write(colormcu->usb_cmd, DEV_CMD_IMU_STREAM_STOP, NULL, 0, NULL, 0) returned failure in colormcu_imu_stop_streaming()
[image]
[image]
[image]

DEPTHKIT FAULT

[2020-07-30 15:09:40.794] [error] [t=8188] ..\src\usbcommand\usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 0
[2020-07-30 15:09:40.797] [error] [t=8188] ..\src\depth_mcu\depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2020-07-30 15:09:40.797] [error] [t=8188] ..\src\sdk\k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
[... 9 lines cut ...]
  Build Config:             Production
  Certificate Type:         Microsoft

The Firmware was updated last week, and there has been a release since, but now every time I try to update it gives me one of two errors - 

EITHER

[2020-07-30 17:47:13.697] [critical] [t=4780] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (366): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2020-07-30 17:47:13.698] [error] [t=4780] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth_mcu\depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2020-07-30 17:47:13.698] [error] [t=4780] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\firmware\firmware.c (128): depthmcu_create(device_index, &firmware->depthmcu) returned failure in firmware_create()

OR...

Please wait, updating device firmware. Don't unplug the device. This operation can take a few minutes...
[2020-07-30 17:56:54.666] [error] [t=6892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (689): usb_cmd_io(). Error calling libusb_bulk_transfer for tx, result:LIBUSB_ERROR_PIPE
[2020-07-30 17:56:54.668] [error] [t=6892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (900): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, &cmd_status) returned failure in usb_cmd_write()
[... cut ...]
[2020-07-30 17:56:54.668] [error] [t=6892] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\firmware\firmware.c (255): depthmcu_download_firmware(firmware->depthmcu, pFirmwareBuffer, firmwareSize) returned failure in firmware_download()
ERROR: Downloading the firmware failed! 1

I did get past this before last week but cant remember. It should be able to handle the Azure Kinect, and has until now. I havent tried factory reset on azure as i know you are limited to one.  Any help please, I have invested so much time and money in this kit only to be stumped now. Thanks in advance!

Computer specs below - 

[image]

qm13, 2020-07-30:
@RobbyGraham looking at the logs we suspect one of two issues:
1. Your USB controller is not one of the recommended controllers. See "USB3 host controller combability" on this page https://docs.microsoft.com/en-us/azure/Kinect-dk/troubleshooting
2. Your USB cable is faulty.
The depth engine uses minimal GPU. Unless you have a very poor GPU or are running other GPU intensive tasks you should very seldom see the "processing to slow" message.

RobbyGraham, 2020-08-01:
> @RobbyGraham looking at the logs we suspect one of two issues:
>
> 1. Your USB controller is not one of the recommended controllers. See "USB3 host controller combability" on this page [...]

Thank you so much. 

The cable was as included with the Azure Kinect, I can buy another and test tomorrow. In the meantime, these are the drivers for the USB hub on my computer - They are intel which should be fine with the Azure Kinect? Have updated all drivers etc for the USB hub.

[image]

The Computer is new and you can see from the specs well within the capacity of dealing with the data according to the specs outlined by Microsoft.

Will try the new cable but are those the only options... Cable was new out the box, if there are any other potential causes or ways to diagnose please do let me know.

RobbyGraham, 2020-08-01:
> @RobbyGraham looking at the logs we suspect one of two issues:
>
> 1. Your USB controller is not one of the recommended controllers. See "USB3 host controller combability" on this page [...]

Hi QM13,

I have replaced the cable and updated all required drivers for USB, uninstalled and re-installed all drivers, and neither has worked. 

Can you re-instate the triage needed element and if anyone else has any ideas as always so appreciated!

wes-b, 2020-08-10:
Do you have another PC you can test the Kinect on? From info you shared above it sounds like your USB Host controller should work. However the errors in your post indicate the controller is not able to keep up. If you have another PC you can try on, then you can determine if the issue is PC related or Kinect related. 

> depth_engine_thread(). Depth image processing is too slow at 35ms (this may be transient).

This means your graphic card is not meeting the needed performance.

> usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth 

This means your PC can not keep up with the performance demands of USB. You might be able to get it to work if you turn off one of the camera's for run at a lower resolution or lower FPS setting.

RobbyGraham, 2020-08-10:
> Do you have another PC you can test the Kinect on? From info you shared above it sounds like your USB Host controller should work. However the errors in your post indicate the controller is not able [...]

Thanks so much for chipping in Wes... I dont have another PC I bought this gaming laptop specifically for the Aure Kinect, as was well within specs... Really frustrating as I had it consistently on for a good hour earlier with Notch, only to come back to it tonight and it is failing. 

Im now getting these errors 

[ warning  ] : is_fw_version_compatable(). Firmware version for RGB is 1.6.102. Consider upgrading to 1.6.110 or newer.
[ warning  ] : is_fw_version_compatable(). Firmware version for Depth is 1.6.75. Consider upgrading to 1.6.79 or newer.
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 39ms (this may be transient).
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000004 unexpected
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_ERROR
[ warning  ] : imu_capture_ready(). A streaming IMU transfer failed
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_STALL
[ error    ] : imu_get_sample(device->imu, imu_sample, timeout_in_ms) returned failure in k4a_device_get_imu_sample()
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_ERROR
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_ERROR
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_ERROR
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_ERROR
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_ERROR
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_ERROR
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_ERROR
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 41ms (this may be transient).
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000001 unexpected
[ error    ] : OnReadSample(). Pipeline propagate error to callback: 0xc00d3ea2
[ warning  ] : capturesync_add_capture(). Capture Error Detected, Color 
[ error    ] : capturesync_get_capture(device->capturesync, capture_handle, timeout_in_ms) returned failure in k4a_device_get_capture()
[ error    ] : usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_IO
[ error    ] : usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, &cmd_status) returned failure in usb_cmd_write()
[ error    ] : usb_cmd_write(colormcu->usb_cmd, DEV_CMD_IMU_STREAM_STOP, NULL, 0, NULL, 0) returned failure in colormcu_imu_stop_streaming()
[ warning  ] : color_capture_available(). A streaming color transfer failed
[ warning  ] : color_capture_available(). A streaming color transfer failed
[ error    ] : Stop(). Failed to request flush for stop: 0xc00d3ea2
[ error    ] : usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_IO
[ error    ] : usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, cmd_status) returned failure in usb_cmd_write_with_status()
[ error    ] : usb_cmd_write_with_status(depthmcu->usb_cmd, DEV_CMD_DEPTH_STREAM_STOP, NULL, 0, NULL, 0, &cmd_status) returned failure in depthmcu_depth_stop_streaming()
[ error    ] : usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_IO
[ error    ] : usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, cmd_status) returned failure in usb_cmd_write_with_status()
[ error    ] : usb_cmd_write_with_status(depthmcu->usb_cmd, DEV_CMD_DEPTH_STOP, NULL, 0, NULL, 0, &cmd_status) returned failure in depthmcu_depth_stop_streaming()

Followed by this error - always a malfunction

[image]

I do need to update firmware but real chicken and egg, did factory reset and tentatively worked so afraid to do anything - really at wits end. Again any suggestions appreciated - in meantime will seeif i can get another pc to test. is there anything i can do about processing or about usb controller ?

wes-b, 2020-08-10:
In this thread, gaming PC's would work until a 3rd party package was downloaded and installed in the background. Your errors seem more like a USB controller / cable / device  issue rather than that, but sharing incase it helps. 

Make sure you try all USB 3.X ports as they are not always equal.

closed 2020-08-29 by qm13

haeginh, 2021-06-09:
I have got exactly same problem! But the difference is that mine used to work well and that another device still works okay with the exactly same setting... One trick you may be able to use is to lower your fps to 15 fps or 5 fps.. In my case it worked for low fps.