microsoft/Azure-Kinect-Sensor-SDK #1303: Misreading Temperature Data

HungrySkeleton, 2020-07-30:
Hello SDK Staff,

I was wondering why the k4a_get_temperature_c reports a different floating point value compared against the kinect viewer's internal temperature sensor reading. Is there something I am doing wrong. I am passing in the capture handle to the k4a_get_temperature_c and it retrieve me a value at 27.5 is that the same reading as the internal sensor temperature shown in the kinectviewer. See attach images below.

[image]
[image]
[image]

qm13, 2020-07-30:
@HungrySkeleton the temperature is measured every time an image capture is taken. Steady state should be somewhere between 25 and 29 degC. It can take some time to reach steady state. It is not clear to me the time difference between the k2aviewer measure and your code.

SeanStephensenAIRM, 2020-08-07:
Can this be reopened? It doesn't seem to be solved. Using the same API call, I get similar results. @qm13, can you clarify what you mean by steady state? Using the k4a viewer, my camera temp reports around 21C when first opened. After streaming for an hour, it has reached steady state around 27.5C as you predicted. Using the API calls to continuously capture for about an hour, the camera's reported temp (which began at around 32C) stabilizes at around 37C. Are we doing something wrong, or is this a bug? In both of my tests, the same camera was used on the same computer. The camera was allowed to cool down to room temperature between tests.

SeanStephensenAIRM, 2020-08-07:
To elaborate, both my K4A Viewer and my NuGet package are SDK v1.4.0. Below is the minimum reproducible code for which the API "misreads" the internal temperature, based on the sample "first project" code directly from Microsoft, simplified for tidiness. The only addition to the sample code is the temperature call. A while loop is used for steady state monitoring. With this minimum reproducible example, the API reports an initial temperature of 33, and increases to around 37. The K4A viewer still reports low 20s at first and increases to mid 20s over a short test. Switching back and forth between the two applications, with or without a cooldown break, has no effect on these numbers. API always reports low/mid 30s, Viewer always reports low/mid 20s. 

```
int main()
{
    k4a_device_t device = NULL;
    k4a_device_open(K4A_DEVICE_DEFAULT, &device);

    k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
    config.camera_fps = K4A_FRAMES_PER_SECOND_15;
    config.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;
    config.color_resolution = K4A_COLOR_RESOLUTION_3072P;
    config.depth_mode = K4A_DEPTH_MODE_NFOV_2X2BINNED;

    k4a_device_start_cameras(device, &config);

    while (1)
    {
        k4a_capture_t capture;
        k4a_device_get_capture(device, &capture, K4A_WAIT_INFINITE);
        float temp = k4a_capture_get_temperature_c(capture);
        cout << temp << endl;
        k4a_capture_release(capture);
    }

    k4a_device_stop_cameras(device);
    k4a_device_close(device);

    return 0;
}
```

qm13, 2020-08-08:
@SeanStephensenAIRM the Azure Kinect has two temperature sensors - one for the IMU and one for the depth sensor. The IMU should be ~27 degC and the depth sensor should be ~37 degC.

[image]

The image is for my Azure Kinect running for ~4 hours in a 25 degC room.

SeanStephensenAIRM, 2020-08-09:
Thanks for the attention, @qm13. Are you able to comment on any functional differences between the two sensors? Am I correct to assume that the IR emitter would be a cause of the higher temp for the depth sensor vs the IMU? I'll do some of my own testing, but can you think of any reason why one temperature sensor would be preferred for a failsafe feature based on internal temperature monitoring? I.E. would you expect both temperature sensors to change uniformly as the external temperature changes (from ambient temp or direct sunlight)?

qm13, 2020-08-29:
@SeanStephensenAIRM Sorry for the tardy response. Yes, you should pay attention to the IR emitter temperature. I would not expect them to change uniformly. The IR emitters are in the front of the device, the IMU is in the rear of the device, and they are separated by the cooling chunnel.

kramakr4, 2020-09-08:
@qm13 Is there any active/passive cooling inside the Kinect? Also, when I leave the camera switched off long enough and then start it up the IMU temperature reads about 6-7 deg C below the room temperature. How is that possible? Is there any bias added to this particular sensor? Or is there any cooling device inside that keeps it so cool?

qm13, 2020-09-11:
@kramakr4 there is no active cooling. The chunnel between the front and back of the device supplies passive cooling.

Regarding the IMU temperature. I observe a similar delta (mine is more like ~14 deg C lower). Investigating why.

kramakr4, 2020-09-11:
If you observe the IMU temperature (blue) when ever I stop recording (every 10000 seconds) and restart, the temperature is higher and then again starts to reduce. Something seems off about this.

[image]

qm13, 2020-09-11:
I have verified that we do nothing to the temperature value in the Sensor SDK. I have forwarded the issue to the hardware / firmware team.

closed 2021-04-13 by qm13