microsoft/Azure-Kinect-Sensor-SDK #1304: Cuda error when creating tracker using C# wrapper on Linux

chitsaw, 2020-08-02:
**Describe the bug**

We are using the C# wrapper from the Microsoft.Azure.Kinect.BodyTracking NuGet to write a simple body tracking test application. The application is .NET Core and runs fine on Windows, but on Linux (with the same hardware), we get the following CUDA exception when calling `Tracker.Create(...)`:

```
/onnxruntime_src/onnxruntime/core/providers/cuda/cuda_call.cc:97 bool onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE = cudaError; bool THRW = true] /onnxruntime_src/onnxruntime/core/providers/cuda/cuda_call.cc:91 bool onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE = cudaError; bool THRW = true] CUDA failure 10: invalid device ordinal ; GPU=0 ; hostname=asi-si-01 ; expr=cudaSetDevice(device_id_); 
Stacktrace:

Stacktrace:

[2020-08-02 00:48:58.883] [error] [t=9087] [K4ABT] /home/vsts/work/1/s/src/TrackerHost/TrackerHost.cpp (157): Create(). /onnxruntime_src/onnxruntime/core/providers/cuda/cuda_call.cc:97 bool onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE = cudaError; bool THRW = true] /onnxruntime_src/onnxruntime/core/providers/cuda/cuda_call.cc:91 bool onnxruntime::CudaCall(ERRTYPE, const char*, const char*, ERRTYPE, const char*) [with ERRTYPE = cudaError; bool THRW = true] CUDA failure 10: invalid device ordinal ; GPU=0 ; hostname=asi-si-01 ; expr=cudaSetDevice(device_id_); 
Stacktrace:

Stacktrace:

[2020-08-02 00:48:58.883] [error] [t=9087] [K4ABT] /home/vsts/work/1/s/src/sdk/k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()
```

We get this error when attempting to create the tracker in both GPU as well as CPU modes. However, the C++ [simple_sample](https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/simple_sample) runs just fine, as does the k4abt_simple_3d_viewer that shipped with the SDK.

It appears to be an issue only when creating the tracker using the C# SDK, and only on Linux.

**To Reproduce**
1. Create a tracker using the `Tracker.Create` C# API on Linux.
2. Method throws the above-mentioned exception.
3. The following simple test program illustrates the issue. Build it using `dotnet build /property:Platform=x64` on Linux and run the resulting executable to reproduce the error:

[AzureKinectTest.zip](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/5011837/AzureKinectTest.zip)

**Expected behavior**

The `Tracker.Create` method should succeed and the application should run.

**Logs**

[log.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/5011828/log.txt)

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: Linux Ubuntu 18.04.4 LTS (Bionic Beaver)
 - SDK Version:  
    libk4a1.3, 1.3.0
    libk4a1.3-dev, 1.3.0
    libk4abt1.0, 1.0.0
    libk4abt1.0-dev 1.0.0
 - Firmware version:  
    RGB camera: 1.6.102
    Depth camera: 1.6.75
    Depth config file: 6109.7
    Audio firmware: 1.6.14

**Additional context**

bearpaw, 2020-10-23:
I have the same issue by running `k4abt_simple_3d_viewer` after I upgraded the nvidia driver to  455.23.05.

KevinHwangAi, 2020-12-17:
I also had the same problem on Ubuntu 18.04. How is the problem solved now?

qm13, 2021-03-24:
Duplicate of #1207. Linux currently does not support C#.

closed 2021-03-24 by qm13