microsoft/Azure-Kinect-Sensor-SDK #1313: Depth error is in the vertical direction of the horizontal plane

monkey-king1, 2020-08-10:
I'm trying to find out the depth accuracy error.I make sure that the color camera faces the desktop.I find that  the depth data is off by 1.5cm in the vertical direction.I wonder how I can make sure the depth data is right.Or maybe the camera has something wrong? The camera is 79.1cm away from the desktop,and the minest depth data is 78.7cm and the maxest depth data is 80.2cm
 
[image]

RoseFlunder, 2020-08-10:
I dont know if I understand your problem correctly, but did you consider that the color camera and depth camera don't have the same angle?
The depth camera is tilted by 6 degrees so the coordinate systems are not the same.

https://docs.microsoft.com/en-Us/azure/kinect-dk/coordinate-systems#depth-and-color-camera

monkey-king1, 2020-08-10:
> I dont know if I understand your problem correctly, but did you consider that the color camera and depth camera don't have the same angle?
> The depth camera is tilted by 6 degrees so the coordinate [...]
so how can i convert depth data from a depth camera to depth data from a color camera.I'd like to get the depth data from the color camera

RoseFlunder, 2020-08-10:
You can take a look at the transformation functions:
https://docs.microsoft.com/en-us/azure/kinect-dk/use-image-transformation

qm13, 2020-08-10:
The expected error is documented here https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#depth-camera-supported-operating-modes. Per @RoseFlunder you are probably generating multi-path due to the 6 deg angle and setup. The expected error at 79cm should be <12 mm.

closed 2020-08-11 by monkey-king1