microsoft/Azure-Kinect-Sensor-SDK #1319: How to open a Kinect2 camera remotely?

buyanxi, 2020-08-17:
A kinect2 camera is connected to a computer called B.
A computer called A which SSH remote login B.
And then execute the capture image test program which is available in computer B， the following error is reported:

Found 1 connected devices.
Done: open device.
deviceSerialnum:. 000296402512
[2020-08-17 18:07:23.483] [error] [t=29882] ../src/dewrapper/dewrapper.c (154): depth_engine_start_helper(). Depth engine create and initialize failed with error code: 204.
[2020-08-17 18:07:23.483] [error] [t=29882] ../src/dewrapper/dewrapper.c (160): deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper()
[2020-08-17 18:07:23.483] [error] [t=29882] ../src/dewrapper/dewrapper.c (194): depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread()
[2020-08-17 18:07:23.484] [error] [t=29866] ../src/dewrapper/dewrapper.c (552): dewrapper_start(). Depth Engine thread failed to start
[2020-08-17 18:07:23.484] [error] [t=29866] ../src/depth/depth.c (398): dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start()
[2020-08-17 18:07:23.484] [error] [t=29866] ../src/depth_mcu/depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[2020-08-17 18:07:23.484] [error] [t=29866] ../src/depth_mcu/depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[2020-08-17 18:07:23.484] [error] [t=29866] ../src/sdk/k4a.c (895): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()
terminate called after throwing an instance of 'k4a::error'

How to solve the above error????  Thanks!!!

qm13, 2020-08-17:
Dup of #810. Also updated the title from "Kinect2" to "Azure Kinect"

closed 2020-08-17 by qm13

buyanxi, 2020-08-18:
I tried to use " export didplay =: 1" and it worked. However, when I tried to run the test program several times and failed which still reported the same error. Then I tried to use " export didplay =: 0" and  it also worked. I think this is the current computer terminal has changed?

According to the above method, in order to successfully open a Azure Kinect camera, there must be a display terminal. But  the way is not that I want to open a Azure Kinect camera. 

I hope that after the computer which is no display device and connected to the Kinect camera is turned on,  and the program running which will be able to open Kinect camera device and take photos and so on. Is this feasible?? Thanks!