microsoft/Azure-Kinect-Sensor-SDK #1320: Kinect Azure device (ubuntu 16.04) - error [libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=2]

buyanxi, 2020-08-18:
when i have installed the sdk of kinect azure device and execute ./build/bin/k4aviewer, the following error is reported:

libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=2
[2020-08-19 14:58:57.460] [error] [t=3350] ../src/usbcommand/usbcommand.c (748): usb_cmd_io(). Error calling libusb_bulk_transfer for initial tx, result:LIBUSB_ERROR_IO
[2020-08-19 14:58:57.460] [error] [t=3350] ../src/usbcommand/usbcommand.c (805): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2020-08-19 14:58:57.460] [error] [t=3350] ../src/depth_mcu/depth_mcu.c (126): usb_cmd_read(depthmcu->usb_cmd, DEV_CMD_DEPTH_READ_PRODUCT_SN, NULL, 0, (uint8_t *)temp_serial_number, sizeof(temp_serial_number) - 1 , &bytes_read) returned failure in depthmcu_get_serialnum()
[2020-08-19 14:58:57.460] [error] [t=3350] ../src/sdk/k4a.c (144): depthmcu_get_serialnum(device->depthmcu, serial_number, &serial_number_size) != K4A_BUFFER_RESULT_SUCCEEDED returned failure in k4a_device_open()

Azure Kinect Viewer -> Open Device shows that : No available device

How to solve the above error?  Thanks!!

buyanxi, 2020-08-18:
this error should be different from another one https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/485

qm13, 2020-08-24:
Number #485 would be normal response to such issues. The only ubuntu release we officially support is 18.04. We have not tested in 16.04.

closed 2020-08-29 by qm13

hlyf-xs, 2021-11-19:
> Number #485 would be normal response to such issues. The only ubuntu release we officially support is 18.04. We have not tested in 16.04.

I also meet the question in 18.04

bestgaomu, 2023-07-12:
I also meet the question in 20.04.Have you solved it yet?