microsoft/Azure-Kinect-Sensor-SDK #1322: Recording the two sensor temperatures using Windows Terminal/Powershell

kramakr4, 2020-08-18:
Hi,

Please excuse my naive questions. I am new to coding and Kinect ecosystem. I have been recording some video samples using k4arecorder using the following command:

k4arecorder.exe -c OFF -l 2700 -r 5 path\filename.mkv

When I open the file using k4aviewer, the IMU data does not have the two temperature sensors on-board the camera. Is there a way to record the sensor temperatures simultaneously when the camera is recording the video? Or may be record it as a part of the IMU data in the video and then output it on to a .csv or .xlsx file?

I use ffmpeg to convert my .mkv file to .tiff images using the following command:

ffmpeg.exe -i path\filename.mkv -map 0:1 path\depth%05d.tiff

qm13, 2020-08-24:
The k4arecorder does record temperature with the IMU payload and it should be available on playback.

kramakr4, 2020-08-24:
[image]

Here is a screenshot of the recording. I did not turn off imu while recording. I just see the accelerometer and gyroscope data and not the temperature.

kramakr4, 2020-08-24:
Here is a screenshot of the recording. I did not turn off imu while recording. I just see the accelerometer and gyroscope data and not the temperature.

[image]

qm13, 2020-08-29:
@kramakr4 yes, k4aviewer does not show the temperatures from a recording, however they are recorded by k4arecorder with the IMU data. Are you reporting an Enhancement request to k4aviewer or a bug in k4arecorder?

kramakr4, 2020-08-29:
I am more interested in taking out the temperature data on to a .csv or .xlsx file. How can I output the temperature data from a recorded .mkv video with IMU data from k4arecorder?

RoseFlunder, 2020-08-29:
If you are familiar with programming you can use the playback API to iterate over the IMU samples in the recording file and then write them to a csv file.

Some explanation for the playback api can be found here:
https://docs.microsoft.com/en-us/azure/kinect-dk/record-playback-api

kramakr4, 2020-08-31:
Hi! I am not familiar with programming. I have tried to go through the tutorials to set up C++ on Visual Studio and run the following code: 

#include<k4a/k4a.h>
#include <k4a/k4a.hpp>
#include<k4arecord/playback.h>
#include <k4arecord/playback.hpp>

k4a:: playback playback = k4a::playback::open("C:\User\krr\Desktop\test.mkv");

k4a::capture capture;
bool result = playback.get_next_capture(&capture);
if (!result) 
{
	break;
}

playback.close()

But, I am not able to get around to get this working. Also, not sure how to query and write the IMU temperature data. Are there any tutorials to get through this and not just the documentation? 

However, I did try mkvtool software's GUI and found that the IMU data only has 6 channels (3 accelerometer and 3 gyroscope). I did not see any temperature data in that.

kramakr4, 2020-08-31:
I have tried running the default code for playback from this link: https://docs.microsoft.com/en-us/azure/kinect-dk/record-playback-api 

Even that did not work/

RoseFlunder, 2020-09-01:
@qm13 
As kramakr4 pointed out there is indeed no temperature data in the recordings.
Here we can see the code that records the IMU sample:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/src/record/sdk/record.cpp#L675
There we can see that only the acceleromter and gyroscope data are written.

@kramakr4 
You need a "main" method to run the code. There are some complete examples in this repository and in another one, but none matches exactly your needs, but they should help to understand where to run the code.

Nevertheless it seems like the tempreture data is missing in the recordings as you pointed out.

kramakr4, 2020-09-01:
@RoseFlunder 
Thanks for your help! I will look into those examples. And yes, the temperature data is not recorded with the IMU data. 

@qm13 
Is it possible to add this to the k4aviewer and also the IMU data as an update?

qm13, 2020-09-11:
I have confirmed with the developer who wrote the recorder API (he has moved to another team at Microsoft) and he confirmed recording the temp of both the IMU and depth sensor was not a requirement.

kramakr4, 2020-09-11:
Thanks @qm13 !