microsoft/Azure-Kinect-Sensor-SDK #1325: Any c# specific documentation available?

nicobeek, 2020-08-19:
While browsing through the functions at https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/index.html
I noticed the k4a_device_get_color_control_capabilities() function listed under C. In the C# assembly I can only find a Device.GetColorControl and Device.SetColorControl. Are there any other functions not available in C#, and is there a C# specific documentation available? The link only has C and C++ documentation.

Thanks!
Nico

qm13, 2020-08-24:
You can find the C# documentation [here](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/namespace_microsoft_1_1_azure_1_1_kinect_1_1_sensor.html)

rcffc, 2020-10-16:
@qm13 Thanks, it was quite hard to find.

qm13, 2020-12-16:
Added links in troubleshooting section of the documentation.

closed 2020-12-16 by qm13