microsoft/Azure-Kinect-Sensor-SDK #1326: non-GPU depth_image_to_color_camera_custom() fails to write invalid values correctly

diablodale, 2020-08-19:
`transformation_depth_to_color()` does not correctly set the `invalid_custom_value` on the entire custom image. The actual pixels set vary based on the stride.

### Setup

* Found while reviewing the code for `k4a_transformation_depth_image_to_color_camera_custom()` which leads to `transformation_depth_to_color()`
* Kinect Azure SDK b033d2e552456e3f830c9c5e9eede04374723540
* Review the file `src/transformation/rgbz.c`

### Repro

Review this code

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/61951daac782234f4f28322c0904ba1c4702d0ba/src/transformation/rgbz.c#L534-L558

### Results

In the three steps...step one is ok. Steps two (CUSTOM8) and three (CUSTOM16) are errant.

These steps two and three do not take into account the stride of the custom image. It is very possible for the pixels to **not** be contiguous in memory layout. Therefore, the `for()` loops are errant to iterate from 0 to num_pixels in a POD array. Images often have strides which result in **not** contiguous pixels. Therefore,  steps 2 and 3 will not correctly pre-set all the pixels to `context->invalid_value` in these non-contiguous scenarios.

### Expected

All pixels in all stride possibilities to have `context->invalid_value` in their invalid pixels.

One correct approach would be to iterate over the rows in both steps , and then iterate within each row to set the `context->invalid_value`

A additional improvement would be to use `memset()` like the following for step 2 (CUSTOM8)

```
if (context->enable_custom8)
{
    memset(context->transformed_custom_image.data_uint8,
           (uint8_t)context->invalid_value,
           (size_t)(context->transformed_custom_image.descriptor->stride_bytes *
                    context->transformed_custom_image.descriptor->height_pixels));
}
```

### Caution

I also saw some suspicious code below that seems to also not take into account the stride of the image. If this is also errant code, then I suspect there is a chain of errant code throughout this part of the code and I recommend it be comprehensively code reviewed.

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/61951daac782234f4f28322c0904ba1c4702d0ba/src/transformation/rgbz.c#L605-L620

qm13, 2020-09-12:
This happens on the CPU path only. GPU (default) path is ok.

diablodale, 2021-07-18:
Found the same bugs -- not using stride and assuming contiguous memory for buffers in the sibling api `transformation_color_to_depth()`. You can see it is using `idx` to go through the buffer using only `width*height` and assuming there is no relevant stride or gaps at the end of rows.

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/04168adc949c44180c51d90a8b8d34807891ec85/src/transformation/rgbz.c#L896-L905