microsoft/Azure-Kinect-Sensor-SDK #1327: ubuntu 16 v4aviewer can not find devices

Crazysiri, 2020-08-21:
here is the steps:
1. sudo apt-get update && sudo apt-get install -y \
        pkg-config \
        ninja-build \
        doxygen \
        clang \
        gcc-multilib \
        g++-multilib \
        python3 \
        git-lfs \
        nasm \
        cmake \
        libgl1-mesa-dev \
        libsoundio-dev \
        libvulkan-dev \
        libx11-dev \
        libxcursor-dev \
        libxinerama-dev \
        libxrandr-dev \
        libusb-1.0-0-dev \
        libssl-dev \
        libudev-dev \
        mesa-common-dev \
        uuid-dev \
        libopencv-dev

2.(these are failed to install,Error: Unable to locate package libk4a1.3)
sudo apt install libk4a1.3
sudo apt install libk4a1.3-dev
sudo apt install k4a-tools

3.
git clone -b release/1.3.x https://github.com/microsoft/Azure-Kinect-Sensor-SDK.git
     mkdir build && cd build
     cmake .. -GNinja
     ninja
     sudo ninja install

4.
Copy 'scripts/99-k4a.rules' into '/etc/udev/rules.d/'.
Detach and reattach Azure Kinect devices if attached during this process.

5.
run 
Azure-Kinect-Sensor-SDK/build/bin/k4aviewer 
or run
AzureKinectFirmwareTool -l

6. error occurs:

 == Azure Kinect DK Firmware Tool ==
[2020-08-21 12:29:51.950] [critical] [t=8118] ../src/usbcommand/usbcommand.c (366): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2020-08-21 12:29:51.951] [error] [t=8118] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2020-08-21 12:29:51.957] [critical] [t=8118] ../src/usbcommand/usbcommand.c (366): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2020-08-21 12:29:51.957] [error] [t=8118] ../src/color_mcu/color_mcu.c (74): usb_cmd_create(USB_DEVICE_COLOR_IMU_PROCESSOR, device_index, NULL, &colormcu->usb_cmd) returned failure in colormcu_create_by_index()
ERROR: Unable to find a device
Found 0 connected devices:

what else should I do?

Crazysiri, 2020-08-21:
could anybody helps? thanks

Crazysiri, 2020-08-21:
I've fixed it, I forgot to put libdepthengine.so.2.0 to /usr/lib/x86_64-linux-gnu/ folder

closed 2020-08-21 by Crazysiri

lulindeng, 2021-03-12:
> here is the steps:
> 
>     1. sudo apt-get update && sudo apt-get install -y 
>        pkg-config 
>        ninja-build 
>        doxygen 
>        clang 
>        gcc-multilib 
>        g++-multilib 
>        python3 
>        git-lfs 
>        nasm 
>        cmake 
>        libgl1-mesa-dev 
>        libsoundio-dev 
>        libvulkan-dev 
>        libx11-dev 
>        libxcursor-dev 
>        libxinerama-dev 
>        libxrandr-dev 
>        libusb-1.0-0-dev 
>        libssl-dev 
>        libudev-dev 
>        mesa-common-dev 
>        uuid-dev 
>        libopencv-dev
> 
> 
> 2.(these are failed to install,Error: Unable to locate package libk4a1.3)
> sudo apt install libk4a1.3
> sudo apt install libk4a1.3-dev
> sudo apt install k4a-tools
> 
> 
> git clone -b release/1.3.x https://github.com/microsoft/Azure-Kinect-Sensor-SDK.git
> mkdir build && cd build
> cmake .. -GNinja
> ninja
> sudo ninja install
> 
> 
> Copy 'scripts/99-k4a.rules' into '/etc/udev/rules.d/'.
> Detach and reattach Azure Kinect devices if attached during this process.
> 
> 
> run
> Azure-Kinect-Sensor-SDK/build/bin/k4aviewer
> or run
> AzureKinectFirmwareTool -l
> 
>     1. error occurs:
> 
> 
> == Azure Kinect DK Firmware Tool ==
> [2020-08-21 12:29:51.950] [critical] [t=8118] ../src/usbcommand/usbcommand.c (366): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
> [2020-08-21 12:29:51.951] [error] [t=8118] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
> [2020-08-21 12:29:51.957] [critical] [t=8118] ../src/usbcommand/usbcommand.c (366): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
> [2020-08-21 12:29:51.957] [error] [t=8118] ../src/color_mcu/color_mcu.c (74): usb_cmd_create(USB_DEVICE_COLOR_IMU_PROCESSOR, device_index, NULL, &colormcu->usb_cmd) returned failure in colormcu_create_by_index()
> ERROR: Unable to find a device
> Found 0 connected devices:
> 
> what else should I do?

Hi,

I followed your steps to build sdk on ubuntu 16.04, but when I run 'ninja', I encountered the error
`[39/151] Building CXX object examples/green_screen/CMakeFiles/green_screen.dir/main.cpp.o
FAILED: /usr/bin/c++   -I/usr/include/opencv -Isrc/sdk/include -I../include -O2 -g -DNDEBUG -fPIE -fvisibility=hidden -fvisibility-inlines-hidden   -Wall -Wextra -Wno-missing-field-initializers -Werror -std=gnu++11 -MD -MT examples/green_screen/CMakeFiles/green_screen.dir/main.cpp.o -MF examples/green_screen/CMakeFiles/green_screen.dir/main.cpp.o.d -o examples/green_screen/CMakeFiles/green_screen.dir/main.cpp.o -c ../examples/green_screen/main.cpp
../examples/green_screen/main.cpp: In function ‘Transformation stereo_calibration(const k4a::calibration&, const k4a::calibration&, const std::vector<std::vector<cv::Point_<float> > >&, const std::vector<std::vector<cv::Point_<float> > >&, const Size&, const Size&, float)’:
../examples/green_screen/main.cpp:505:116: error: could not convert ‘16648’ from ‘int’ to ‘cv::TermCriteria’
                                        cv::CALIB_FIX_INTRINSIC | cv::CALIB_RATIONAL_MODEL | cv::CALIB_CB_FAST_CHECK);
                                                                                                                    ^
[39/151] Building CXX object tests/DepthTests/UnitTest/CMakeFiles/depth_ut.dir/depth_ut.cpp.o
ninja: build stopped: subcommand failed.
`

Seems like it is caused by the opencv, Do you have idea about this error? I installed the opencv by `sudo apt install libopencv-dev`.

Thanks a lot for your reply!