microsoft/Azure-Kinect-Sensor-SDK #1333: Confusion about avoiding interference and offset setting in document

seigeweapon, 2020-08-28:
**Describe the bug**

I'm reading [https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync](url) , and got confused about how to set correct offset. In the _Avoiding Interference_ section, I read

>To prevent the lasers from interfering with one another, the camera captures should be offset from one another by 160μs or more.
...
For each depth camera capture, the laser turns on nine times and is active for only 125μs each time. The laser is then idle for either 14505μs or 23905μs

To me, the 2nd statement seems to imply the laser pulse in one exposure period is like this:
[image]

But then it's conflicting to the 1st statement. The basic idea of avoiding interference is avoiding laser overlaping in temporal space, right? If so, and the pulse is like what I draw above, an offset of 160us is obviousely not enough.

**To Reproduce**

Document
**Expected behavior**

Make clear of the doc
**Logs**

None
**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04
 - SDK Version:  1.4.1
 - Firmware version:  

**Additional context**

None

qm13, 2020-08-31:
The correct documentation link is [this](https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync#avoiding-interference-between-multiple-depth-cameras). Note also that the document references the [raw timing](https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification#depth-sensor-raw-timing) for the 4 operating modes.
The actual pulse width is 125us however we state 160us to provide for some leeway.

Taking NFOV unbinned as an example, each 125us pulse is followed by 1450us idle. Totaling these up - (9 x 125) + (8 x 1450) - yields the exposure time of 12.8ms. The closet you can interleave exposure of 2 devices is to have the first pulse of the second camera to full in the first idle period of the first camera. The delay between the first and second cameras could be as little as 125us (the width of a pulse) however we recommend some leeway hence the 160us. Given 160us you can interleave the exposure periods of 10 cameras.

There is also the time between the end of the first exposure period - 12.8ms for our NFOV unbinned example - and the start of the second exposure period. This time is dependent on the frame rate. For our NFOV unbinned example running at 30fps this would be - 33ms - 12.8ms - 20.2ms. You could for example mount 2 cameras on top of each other and given suitable calibration create a pseudo 60fps setup by delaying the second camera by 16.5ms.

qm13, 2020-12-16:
Updated documentation

closed 2020-12-16 by qm13