microsoft/Azure-Kinect-Sensor-SDK #1334: Calculating distance of an object from the camera (using IR camera)

jovinteng, 2020-08-31:
I'm using a retroreflective ball marker as my object and have so far been able to detect it using the Kinect Azure's IR camera and with the help of the Hough circles transform function from OpenCV. Doing so provided me with the centroid's x and y-coordinates in the IR domain (which I assume is the same as the depth camera domain). I want to transform these 2D pixel coordinates into 3D coordinates to then be able to detect how far away (in mm) the marker is from the camera. I've looked around the forums and haven't really found anything that works for me so far... If anyone could provide some guidance and advice that'd be much appreciated! Thank you.

qm13, 2020-08-31:
The active IR capture (your can also run the camera in passive mode) and the depth capture are from the same sensor. There is no transformation between captures. You can take the x and y value from your centroid calculation and directly read the z value from the depth capture. The z value is mm from the camera and has a max error of 11mm + 0.1% of range.

jovinteng, 2020-09-01:
@qm13 I see... So there's no need to transform the 2D depth capture to a 3D point cloud first? Also, is there a way to make the depth measurement more accurate because 11mm is a big margin of error.

qm13, 2020-09-11:
@jovinteng correct.

There is no way to make the depth error smaller for a single Azure Kinect. Some developers are exploring using multiple Azure Kinect devices to reduce the error.

closed 2020-09-14 by jovinteng