microsoft/Azure-Kinect-Sensor-SDK #1335: Error LibUSB transfer status

ArmandoNN113, 2020-08-31:
**Describe the bug**
When I turn on the camera (depth and color) in the k4aviewer, I get an error message of the cameras timed out and the following errors in the log:
```
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 89ms (this may be transient).
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    268533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    301866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    335166 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    368533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    435200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    468533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    501866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    535177 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    568533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    601855 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    635200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    668533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    701866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    735188 type:Depth
```
Depth Camera Only

When I turn on the depth camera only and at 5-15 fps, the camera works ok, but gives the following log when set to 30 fps:
```
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 84ms (this may be transient).
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000004 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_STALL
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000004 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_STALL
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000004 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_STALL
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000004 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_STALL
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000004 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_STALL
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000004 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_STALL
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000004 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_STALL
[ error    ] : usb_cmd_libusb_cb(). LibUSB transfer status of 00000004 unexpected
[ error    ] : usb_cmd_libusb_cb(). Error LIBUSB transfer failed, result:LIBUSB_TRANSFER_STALL
[ error    ] : usb_cmd_io(). Error calling libusb_bulk_transfer for status, result:LIBUSB_ERROR_PIPE
[ error    ] : usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, cmd_status) returned failure in usb_cmd_write_with_status()
[ error    ] : usb_cmd_write_with_status(depthmcu->usb_cmd, DEV_CMD_DEPTH_STREAM_STOP, NULL, 0, NULL, 0, &cmd_status) returned failure in depthmcu_depth_stop_streaming()
[ error    ] : usb_cmd_io(). Error calling libusb_bulk_transfer for status, result:LIBUSB_ERROR_PIPE
[ error    ] : usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, NULL, 0, p_data, data_size, NULL, cmd_status) returned failure in usb_cmd_write_with_status()
[ error    ] : usb_cmd_write_with_status(depthmcu->usb_cmd, DEV_CMD_DEPTH_STOP, NULL, 0, NULL, 0, &cmd_status) returned failure in depthmcu_depth_stop_streaming()
```
Color camera only:
The color camera fails regardless of the settings I input.

**Expected behavior**
I expected the depth and color cameras to work well with the k4aviewer program, to work on a separate Unity project.

**Laptop Specs:**
[image]
[image]

**Additional context**
I have tried this on my office pc and it did work. Unfortunately I am unable to access my office pc at the moment and have my laptop to work with. The error is mostly centered in LibUSB library from the logs. 

Any help is appreciated, thank you!

qm13, 2020-09-03:
Couple of things to check:

Try difference USB port
Remove all other USB devices on that USB controller to make sure another device is not interfering
Test with supplied USB and power cables
Test with an alternative USB cable.

ArmandoNN113, 2020-09-04:
Thanks for the reply @qm13 !!
I tried all of your suggestions, but unfortunately none of them worked. I was able to access another pc and it worked fine. 
I talked with Razer support and we found out that the usb ports were faulty, so this is my laptop not functioning very well. 
I appreciate your time and help!