microsoft/Azure-Kinect-Sensor-SDK #1336: VLC Image is Glitchy

rssuther, 2020-09-01:
I have had my Azure Kinect camera for just under a year and have experienced zero problems until just recently. The image from the visible light camera is glitchy and seems to be a hardware issue. I have reset and reflashed the device firmware but this has not solved the issue. Occasionally the issue does not exist but for the most part, the artifacts in the image are severe. I am not sure if this is a firmware issue, hardware issue, or caused by a Windows update. Is there any knowledge of this problem and possibly a solution? Or is it most likely a hardware failure?

**Steps to reproduce the error:** Basic use of the device as a webcam in teams or in the Asure Kinect Viewer tool.
**Expected results:** Clear and glitch-free images
**Actual results:** Glitchy images
**SDK version:** v1.4.1
**Firmware version:** v1.6.110079014
 - RGB camera firmware:      1.6.110
 - Depth camera firmware:    1.6.79
 - Depth config file:        6109.7
 - Audio firmware:           1.6.14
 - Build Config:             Production

qm13, 2020-09-03:
Couple of things to check:
1. Try difference USB port
2. Remove all other USB devices on that USB controller to make sure another device is not interfering
3. Test with supplied USB and power cables
4. Test with an alternative USB cable.

JosephParkins, 2020-09-14:
I believe I'm having the same issue, coming towards a year with the devices and the error has begun to occur. For me the error occurs for occasional frames, and when captured looks like packet loss. 

**SDK Version v1.2.0**

- RGB camera firmware: 1.6.102
- Depth camera firmware: 1.6.75
- Audio: 1.6.14
- Build Config: Release

I've noticed that when the device is connected directly to the PC with the provided data cable the error does not occur, however when using a cable matters extension cable as recommended at https://docs.depthkit.tv/docs/azure-kinect the error begins to occur.

I've provided a capture of this error for reference.
[image]

closed 2020-11-30 by qm13