microsoft/Azure-Kinect-Sensor-SDK #1338: Kinect drops off USB bus and will not reconnect (Jetson Nano)

fowie, 2020-09-01:
I'm running a project with eight AKs.  Each is connected to a Jetson Nano via the provided USB cable.  Each has its own 5V external power supply.  I've now seen this issue twice where the Kinect can get into an unrecoverable state.  In the most recent case, I was running my code which was accessing the Kinect.  I shut down my code and when I tried to re-start, it could not find a Kinect.  AzureKinectFirmwareTool also reported it could not find a device.  lsusb returned:
`Bus 002 Device 003: ID 045e:097a Microsoft Corp.
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 045e:097e Microsoft Corp.
Bus 001 Device 003: ID 045e:097b Microsoft Corp.
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

for reference, a working linux system with a Kinect connected will return:
`Bus 002 Device 030: ID 045e:097c Microsoft Corp.
Bus 002 Device 031: ID 045e:097d Microsoft Corp.
Bus 002 Device 003: ID 045e:097a Microsoft Corp.
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 045e:097e Microsoft Corp.
Bus 001 Device 003: ID 045e:097b Microsoft Corp.
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

If I look in dmesg, I see a repeating message:
`[     6.613654] usb usb2: usb_suspend_both: status 0
[   14.593487] usb 2-1.4: usb_suspend_both: status 0
[   14.685489] usb 2-1.4: usb_suspend_both: status 0
[   14.733498] usb 2-1: usb_suspend_both: status 0`

This is not resolved by rebooting the nano.  The only solution was to unplug the USB cable from the Kinect and re-connect it, at which point it registered all three Microsoft devices correctly and started working again.

fowie, 2020-09-01:
Using libk4a 1.4 
Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

jmichon, 2020-09-03:
Possible duplicate: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1283

qm13, 2020-09-03:
This is a duplicate of #1283 as @jmichon eludes too. It is a known issue in the firmware which the hardware team in planning a fix.

qm13, 2020-09-11:
Closing as duplicate

closed 2020-09-11 by qm13