microsoft/Azure-Kinect-Sensor-SDK #1341: Depth errors when measuring human skin

crux24, 2020-09-02:
**Background:**
We are trying to measure the height of hands and feet from a top-bottom setup of an Azure Kinect. 

**Problem:** 
When calculating the distance of depth between the highest point of the extremity and the ground, which represents the height, we constantly measure a discrepancy to reality. Specifically around 8-15mm less than the ground truth.

**Measurements:**
We tried measuring from different heights, different Kinect cameras, angles, and objects, and our first guess leads to "material" issue. When putting on socks or measuring boxes, the height seems to be correct. 

**Experiment:** 
So for a quick evaluation we took depth and rgb images from a hand/box, and afterwards from the same object with a piece of paper taped above them. See  clipped images below:
[image]
[image]
[image]
[image]

**Expectation:**
When calculating the difference of the images with and without paper attached we would expect only small variations between the images.

**Results:**
The first image is the difference image of the box. As you can see there are small differences in the area where the paper was placed, but they are in the range of 1 mm.
[image]
Whereas when looking at the hand image, differences in the range of 8mm are visible.
[image]

**Explaination?**
 Is the human skin as reflection "material" unsuitable, so that measurements of plain skin may always lead to errors in this range, or are we missing something else here?
And if it's really the skin, are there ways to correct these errors?

Thanks in Advance!

jasjuang, 2020-09-04:
I agree material severely affects the depth accuracy. I've reported my findings last year here https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/803#issuecomment-570006001.

qm13, 2020-09-06:
Some materials such as human skin and certain plastics (e.g. red LEGO blocks) are fully or partially transmissive of near-IR wavelengths. When the active illumination from the depth sensor reaches the object, rather than being wholly reflected at the air-object interface it penetrates and is scattered, resulting in the mean path length travelled by the illumination being longer than expected (and thus the estimated range being farther away than is actually the case). This can be considered to be a type of multipath interference. There is no trivial way to detect this behavior in the most general case, given the small path length differences involved. This problem is not specific to the Kinect for Azure depth sensor and is a general problem for ToF depth imaging in the near-IR. 

For most applications this offset is not an issue, however if it is problematic and better accuracy is desired there are many possible heuristic solutions. For example, if you are designing for a use case where you start with the specific knowledge that you are going to be tracking a human hand, then you may wish to apply a correction factor to the hand based upon that knowledge.

crux24, 2020-09-08:
@jasjuang, yes thank you, we've read the thread. It actually was the reason, why we ended up searching in this direction.
@qm13 thank you too for your explanation! Sadly the age or skin color of our probands may vary, so that a fixed correction factor may not be enough. But we are currently looking into other workarounds, so thanks again.

closed 2020-09-11 by qm13