microsoft/Azure-Kinect-Sensor-SDK #1342: When can we expect body tracking on AMD64?

anselanza, 2020-09-04:
> This version (1.4.0) does not support body tracking sdk on Ubuntu 18.04.
> Is there any way to deal with this?

Body Tracking team is also adding ARM support. A new version of BT is coming that will work with 1.4.0.

_Originally posted by @wes-b in https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1093#issuecomment-611617173_

anselanza, 2020-09-04:
Just wondering if there has been any update since this last comment on the issue (that I can find), which was back in April.

Completely appreciate that these things take time, but just wondering if there is an ETA?

I have depth/RGB camera input working perfectly on Nvidia Jetson Xavier NX, but of course body tracking is still officially not supported yet for this platform.

qm13, 2020-09-10:
This is still under development.

anselanza, 2020-09-23:
Yes, understood that it is still in development, but was curious if there was any rough timeline? I don't want to put pressure on developers, but since the last update was 5 months ago, I was wondering if it was something coming soon (in the next month) or still half a year away?

aspioupiou, 2020-09-30:
@qm13 Hi !
Could you please give some information about BodyTracking linux support as well ? I need to know if the release might happen soon or not so we can device if w need to create something ourself for body tracking support on linux platform

anselanza, 2020-10-01:
> Could you please give some information about BodyTracking linux support as well ? I need to know if the release might happen soon or not so we can device if w need to create something ourself for body tracking support on linux platform

As far as I know, Body Tracking works just fine on Linux! See https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download#linux-installation-instructions

But currently it only works for x64 architecture. Whereas I would like to try running this on a Jetson Xavier, which uses ARM64.

weidongshao, 2020-12-30:
@qm13 
Any update on timeline of the body-tracking on ARM? 
Is there anything the open source developers can help on the porting efforts? 

https://developer.nvidia.com/blog/announcing-onnx-runtime-for-jetson/

flugi, 2021-02-26:
Any news on ARM64 support for Body Tracking? I have read somewhere an estimation that Xavier NX will be capable, is there any information about this regard?

carloshgaray, 2021-03-12:
+1 for status update. Is this something perhaps the community could help contribute? If so do you have any pointers on how to go about doing this for folks staring out?

Hoping to get body tracking working on nvidia xavier

carloshgaray, 2021-04-21:
Any news on ARM64 support for Body Tracking?

qm13, 2021-07-08:
Unfortunately we have encountered some sever issues getting body tracking running performantly on ARM64.

tank104, 2021-07-08:
@qm13 does that mean you are pausing it, or just working through the issues, and will release soon?

burakaksoy, 2021-11-04:
Still hopefully waiting for the support..

qm13, 2022-05-17:
Duplicate of #871.

closed 2022-05-17 by qm13