microsoft/Azure-Kinect-Sensor-SDK #1343: body tracking support multiple ONNX execution providers (e.g. DirectML)

diablodale, 2020-09-05:
Some of my customers are looking elsewhere for new tracking sensors because these customers have Intel and AMD GPUs.
Azure Kinect Body Tracking is Nvidia only.

Body Tracking 1.0.1 only supports the default CPU + defacto GPU (Nvidia CUDA) ONNX execution providers. That's my guess.
The former naturally doesn't perform, the latter is vendor lock-in. Both decrease our sales and services.

### Request

I request Body Tracking enable a wider selection of execution providers.
ONNX supports 15 providers https://github.com/microsoft/onnxruntime#supported-accelerators

The `DirectML` provider is a good candidate for exploration; could allow acceleration for multiple GPU vendors across multiple generations of GPUs on Windows. https://github.com/microsoft/onnxruntime/blob/master/docs/execution_providers/DirectML-ExecutionProvider.md. Naturally, each execution provider has a different set of capabilities.

### Alternatives

None. Body tracking is closed source and I am unaware of published APIs which can externally modify Body Tracking execution behavior. If something *is* possible, please update this issue and I will try it. I can rebuild the ONNX runtime with whatever options are suggested.

### Related

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1016 and https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1152 and https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1154

qm13, 2020-09-10:
This is under development.

Jeremboo, 2021-03-12:
Hi, do you have any release date in mind for the ADM GPU support?

Thanks in advance!

qm13, 2021-03-22:
Fixed in Body Tracking SDK v1.1.0

closed 2021-03-22 by qm13