microsoft/Azure-Kinect-Sensor-SDK #1344: ssh connection to jetson nano -  depth_engine problems

alitokur, 2020-09-09:
I have 2 machines, one of them is an Ubuntu 18.04 desktop computer and the other is a Jetson Nano. I coded a ROS publisher with Azure SDK, and build it on Jetson Nano. So far so good. Tested ./k4aviewer and my publisher code(./k4aRosPublisherForArm) and they are working well. I can capture color and depth images successfully. 

./k4aRosPublisherForArm
 
[image]

and from the ROS side, everything is ok.

[image]

Now, my main purpose is using jetson nano with no monitor. But before that, both my devices are connected to the monitors for testing. And I'm using ssh to connect to the jetson nano from my desktop. When I try to run my publisher binary, it fails. 

To Reproduce

1. install ros  & azure sdk
2. build azure sdk and putting my publisher code to build/bin
3. run ./k4ARosPublisherForArm

**Expected behavior**
My publisher node working on jetson nano, it should also work when connected remotely to jetson nano.

**Logs and Error Messages**
```

../src/de wrapper/wrapper.c (154): depth engine_start_helper(). Depth engine create and initialize failed with error code: 204.
...
(all logs will add later)

```
inside my header file

```
 switch (k4a_device_get_capture(device_, &capture, TIMEOUT_IN_MS))
		{
		case K4A_WAIT_RESULT_SUCCEEDED:
			break;
		case K4A_WAIT_RESULT_TIMEOUT:
			printf("Timed out waiting for a capture\n");
			return;
		case K4A_WAIT_RESULT_FAILED:
			printf("Failed to read a capture\n");
			return;
		}
```
I get "Failed to read a capture"

**Desktop ** Ubuntu 18.04
 **Jetson Nano** 18.04

**Additional context**
Similar issues have been created before. #810 But I could not find any working solution.

wes-b, 2020-09-10:
See this comment in particular from https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/810#issuecomment-679947486

chethanab16, 2020-09-10:
Hi,

Is Azure-Kinect-Sensor-SDK supported on Lattepanda (Windows OS) , Raspberry Pi 4 (Windows ARM OS) and Intel Compute Stick ?
Does Microsoft hololense azure kinect is same as new azure kinect camera?
thank you in advance

closed 2020-09-10 by alitokur