microsoft/Azure-Kinect-Sensor-SDK #1345: Getting 2D Depth Images from point cloud .ply

fatihoezdemir, 2020-09-10:
Hello ,
I didn't want to have diamond shaped 2D depth images so I used "/transformation_example playback" to mitigate these shaped pictures and I got  .ply files for the pictures
I want to get 2d depth mages from the point cloud .ply files. 

I appreciate your help

qm13, 2020-09-10:
Are you referring to the aperture shape? e.g. https://docs.microsoft.com/en-us/azure/Kinect-dk/media/concepts/depth-camera-depth-ir.png

fatihoezdemir, 2020-09-10:
Yes

fatihoezdemir, 2020-09-10:
My goal is to get lossless rectangle shaped depth images from a recording, not the original aperture shape

fatihoezdemir, 2020-09-11:
I realized that there is a much better way to get transformed images, so this issue is closed. I have another issue regarding my new solution whrere I will open a new issue:)

closed 2020-09-11 by fatihoezdemir