microsoft/Azure-Kinect-Sensor-SDK #1347: Can't record RGB camera captures due to "full queue" error.

Groulaxx, 2020-09-11:
**Describe the bug**

Can't record RGB camera captures due to "full queue" error.
Using the cpp code i linked, the code shows the error for each frame captured but still goes to the end and output a mkv file which is readable with Kinect viewer, but when I read the file I only have Depth images.
The error comes after I called the "k4a_record_create() function and for each k4a_device_get_capture() call. (Example : if I record 200 frames, I get 200 times the error)

**To Reproduce**

Just run my cpp code from this file.
[My_cpp_code.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/5207402/My_cpp_code.txt)

**Expected behavior**

I think that I should get some captures from RGB camera and record some coloured captures from the camera, and see the coloured captures when I open the .mkv file with Azure Kinect Viewer 1.4.1 .

**Logs**

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 
 - SDK Version:  1.4.1 
 - Firmware version:  RGB Cam 1.6.110 - Depth Cam 1.6.79 - Depth config 6109.7 - Audio 1.6.14 
 - Graphic card : Nvidia Quadro M2000M
 - Proc : Intel core i7-6700HQ 2.6GHz

**Additional context**

I don't have any error messages when I open the device using Kinect viewer (not a record) and I can see the color camera stream.
I saw the same error report in few other google pages or issues but none of them helped me fix the problem by now.

RoseFlunder, 2020-09-11:
I have only glanced over your code but it looks like you need too much time for each Iteration of the processing loop. Try to remove the "sleep" calls in the loop like in the "VERIFY" macro.
The buffer from the kinect fills up with data and needs to processed fast enough from your app.

Groulaxx, 2020-09-14:
I tried to remove all my "sleep" calls but it doesn't change anything. I tested the code on other computers and it actually worked well, is it possible that the problem comes from the GPU (which is a bit old) or any other too old hardware ?

qm13, 2020-09-14:
Recording is impacted by disk speed and RAM availability. Try reducing depth resolution, color resolution and FPS. If it is capable of recording at lower settings then it is a PC performance issue.

Groulaxx, 2020-09-15:
I tried multiple configurations and this is what happens : 
- When recording both colour and depth, i get the error and no colour inside the file for all fps or resolution settings.
- When recording only depth, it works well for all fps and resolution config.
- When recording only colour, I get the "k4a_device_get_capture()" call return 'timed out!'.
So finally, it looks like that's not PC performance issue ..

Groulaxx, 2020-09-15:
I just tried to run my same code on linux (Ubuntu 18.04) on the same PC and it worked, so maybe this could be a windows driver issue, but after a quick look, all my drivers are already up to date.

FelipeDelgadoR, 2020-09-16:
I have a similar issue with a camera, it was working a couple of days ago and now is just getting this "full queue" issue, can it be related to windows drivers or something?

wes-b, 2020-09-17:
Are there any errors in the log or shown in K4aViewer on Windows? That should help you determine what is failing in the color camera pipeline.

Groulaxx, 2020-09-18:
[image]
  
This is what I get in log but nothing looks weird to me, and the color camera is looking correctly from k4aViewer, only the recording shows the error mentioned in my original post.

closed 2020-11-30 by qm13