microsoft/Azure-Kinect-Sensor-SDK #1350: [ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    307733 type:Color

siraj-haid, 2020-09-14:
**Describe the bug**
When the WFOV(wide field of view) depth configuration is used, the Azure kinect is failing in Kinect viewer. We tried to update the firmware but it did not work. Everytime after opening the kinect viewer for couple of seconds we receive the following errors in the viewer when crashing:

**To Reproduce**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version: RGB camera: 1.6.102
                                Depth camera: 1.6.75

qm13, 2020-09-14:
Please try flashing the device on a different port or different computer. Suspect you have a USB controller issue.

siraj-haid, 2020-09-15:
I tried to flash the device. Then I replaced the USB cable with one we use for our second Kinect camera and it gave the same error  

Also i tried the camera on 3 different computers and still the same error.

Do you have another suggestions?

Thank you.

qm13, 2020-09-16:
@siraj-haid Can you please try to reset the device to factory defaults and then attempt to flash the device. You can find the instructions [here](https://docs.microsoft.com/en-us/azure/Kinect-dk/reset-azure-kinect-dk)

siraj-haid, 2020-09-25:
@qm13  I changed the power cable of the Kinect camera and for now the camera stopped crashing. Thank you for your help.

closed 2020-09-25 by siraj-haid

guss2149, 2021-08-20:
Hello,  I am currently getting these same errors when I use an alternate power source. Our application require a much longer cable than the original one but we've only had success (no errors) when we run a 110V extension cord and use the original power supply and original power cable. Does anyone know of any other way to power the cameras? Our attempted (with no success) setup is using [this cable](https://www.ebay.com/itm/172817859383?_trkparms=ispr%3D1&amdata=enc%3AAQAFAAAAkFCyoA%2BXzDlXqBnbcJoXJfz9lpLxQ048xN3i2kJF0g3LhrYejm5Ob%2FKGbzm6uifaGmcXAJfsvpYOtMRubeXEhM78InKa4IPck8jtn10%2FazMga1CcPfDnP7Foeu3c3071tXqICL5u2k8U70anGcRnDoGledkru4VkzsTW7zO8Q1p9GcfMEfMyHxzjeSoxO%2BgtgA%3D%3D&chn=ps&norover=1&mkevt=1&mkrid=711-117182-37290-0&mkcid=2&itemid=172817859383&targetid=1262376588856&device=c&mktype=pla&googleloc=9021515&poi=&campaignid=12874531651&mkgroupid=123924950842&rlsatarget=pla-1262376588856&abcId=9300535&merchantid=108712654&gclid=EAIaIQobChMIw8X87Oni8AIVC26GCh2usA_5EAQYAiABEgI2xfD_BwE) and with a 3 conductor cable (only using 2 wires) soldered to the leads connected to [this power supply](https://www.deltapsu.com/en/products/din-rail-power-supply/DRS-5V50W1AZ). We have had no success. Has anyone had any success with any alternate power supply solution?

arkader, 2021-11-02:
Hi @siraj-haid 
I'm currently running into the same issue, which power cable are you using now? 
You're not using the one provided in the box anymore? 

Hi @qm13, I've factory reseted the device manually, what do you mean by flashing it? 

Hi @guss2149, have you fixed the issue on your side?

qm13, 2021-11-02:
@arkader https://docs.microsoft.com/en-us/azure/kinect-dk/update-device-firmware