microsoft/Azure-Kinect-Sensor-SDK #1351: playback image transformation error

fatihoezdemir, 2020-09-14:
Azure Kinect Sensor SDK 1.4.1 Ubuntu 18.04 

Hello everyone, I am getting the following the following error when I want to transform the images of a playback with depth_image_to_color_camera:

**[2020-09-14 12:44:03.770] [error] [t=17273] ../src/transformation/transformation.c (693): transformation_depth_image_to_color_camera_custom(). Expect both depth camera and color camera are running to transform depth image to color camera.
[2020-09-14 12:44:03.770] [error] [t=17273] ../src/sdk/k4a.c (1201): transformation_depth_image_to_color_camera_custom(transformation_handle, depth_image_buffer, &depth_image_descriptor, custom_image_buffer, &dummy_descriptor, transformed_depth_image_buffer, &transformed_depth_image_descriptor, transformed_custom_image_buffer, &dummy_descriptor, interpolation_type, invalid_custom_value) returned failure in k4a_transformation_depth_image_to_color_camera()
terminate called after throwing an instance of 'k4a::error'
  what():  Failed to convert depth map to color camera geometry!
Aborted (core dumped)
**

```C++

#include <k4a/k4a.hpp>
#include <k4arecord/playback.hpp>
#include <iostream>
using namespace std;

int main(){
k4a::playback handle = k4a::playback::open("test.mkv");
k4a::capture capture;
k4a::image depth_image;
k4a::image color_image ;
k4a::image transformed_depth_image;
k4a::calibration calibration;
k4a::transformation transformation;
int no_frames{};
transformation = k4a::transformation(calibration);
calibration=handle.get_calibration();	

while(handle.get_next_capture(&capture))
{
//Preocess Capture
    no_frames+=1;

    depth_image = capture.get_depth_image();
    if(!depth_image)
    {
        std::cout << "No depth_image!" << std::endl;
    }
    color_image = capture.get_color_image();

    transformed_depth_image = k4a::image::create(
            K4A_IMAGE_FORMAT_DEPTH16,
            calibration.color_camera_calibration.resolution_width,
            calibration.color_camera_calibration.resolution_height,
            calibration.color_camera_calibration.resolution_width * (int)sizeof(uint16_t));

    transformation.depth_image_to_color_camera(depth_image, &transformed_depth_image);

}

//number of frames
cout<<no_frames<<endl;

```
I am also fine working with C, my only goal is to get transformed images :)

Thanks

RoseFlunder, 2020-09-14:
You pass in an empty calibration into the transformation.
You have two switch these two lines:
```
transformation = k4a::transformation(calibration);
calibration=handle.get_calibration();	
```

fatihoezdemir, 2020-09-14:
Great, I overlooked this,thank you @RoseFlunder .
But somehow, I still do get the same shape of the images(no transformation). Do I need to do another operation?

qm13, 2020-09-14:
Swap these two lines around

```
transformation = k4a::transformation(calibration);
calibration=handle.get_calibration();
```

You are creating the transform with an empty calibration.

fatihoezdemir, 2020-09-14:
Yes , it fixed this problem, but how can I overwrite these transformed images ?Currently, I am getting untransformed images.

closed 2020-09-16 by fatihoezdemir