microsoft/Azure-Kinect-Sensor-SDK #1355: No transformed depht images at NFOV

fatihoezdemir, 2020-09-16:
**Describe the bug**
I am getting transformed depth images at WFOV mode, but the NFOV images are not transformed , so there is no color map geometry.
**Screenshots**
**This is the  transformed depth image at WFOV with the following settings**
[image]
```C++
    device_configuration = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
    device_configuration.color_format             = k4a_image_format_t::K4A_IMAGE_FORMAT_COLOR_BGRA32;
    device_configuration.color_resolution         = k4a_color_resolution_t::K4A_COLOR_RESOLUTION_720P;
    device_configuration.depth_mode               = k4a_depth_mode_t::K4A_DEPTH_MODE_WFOV_2X2BINNED;
    device_configuration.synchronized_images_only = true;
    device_configuration.wired_sync_mode          = k4a_wired_sync_mode_t::K4A_WIRED_SYNC_MODE_STANDALONE;
    device.start_cameras( &device_configuration );

```
**This is the"transformed" depth image at NFOV (it is the same with any binned, unbinned  setting**
)
[image]
```C++
    device_configuration = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
    device_configuration.color_format             = k4a_image_format_t::K4A_IMAGE_FORMAT_COLOR_BGRA32;
    device_configuration.color_resolution         = k4a_color_resolution_t::K4A_COLOR_RESOLUTION_720P;
    device_configuration.depth_mode               = k4a_depth_mode_t::K4A_DEPTH_MODE_NFOV_2X2BINNED;
    device_configuration.synchronized_images_only = true;
    device_configuration.wired_sync_mode          = k4a_wired_sync_mode_t::K4A_WIRED_SYNC_MODE_STANDALONE;
    device.start_cameras( &device_configuration );
```
**To Reproduce**
I am using the following code: https://github.com/UnaNancyOwen/AzureKinectSample/blob/master/sample/cpp/transformation/kinect.cpp from @UnaNancyOwen 

**Desktop (please complete the following information):**
Ubuntu 18.04
 - SDK Version:  1.4

Is this a bug or isn't NFOV supported. If WFOV has worked, I don't see any hindrance with NFOV

Thanks for your help

RoseFlunder, 2020-09-16:
I thought about it and I think the transformation works and is correct in your pictures.
I think transformation does not mean, that you should get a rectangular image after it. Only that the image looks like it was taken from the perspective of the color camera, so its origin/coordinate system is the one from the color camera.

The reason you have a rectangular image at WFOV is because the depth image covers a larger area than the color camera. So every color pixel you will have overlap with the depth image.
At NFOV this is not the case. The depth image is smaller than the color image, so the transformed image can't be rectangular. It would have no depth information at the outer areas.
You can see the overlap of the different modes in the documentation:
https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification#camera-field-of-view

[image]

fatihoezdemir, 2020-09-16:
Thank you  for your answer, it clarified my question!

closed 2020-09-16 by fatihoezdemir