microsoft/Azure-Kinect-Sensor-SDK #1356: Body tracking does not work well

sicheong, 2020-09-17:
**Describe the bug**
When a leg is raised high, body tracking SDK can't recognize that leg.
I think it happens when a foot rises above the waist.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 1903
 - SDK Version:  1.0.1
 - Firmware version:  1.6.110079014

qm13, 2020-09-17:
Thus is a known issue. See #738.

billpottle, 2020-11-12:
@sicheong - We investigated this issue in detail, and found that it's actually prevalent across a number of models - ie, similar issues happen with OpenPose as well. The general consensus was that we would have to train our own model with thousands of data points to include kicks at high speed.

qm13, 2020-11-30:
@billpottle you are correct. Whilst we did capture a bunch of data of people exercising we did not specifically capture data for high kicks. We are currently researching ways for developers to customize the model (a core reason we opted to move to DNN from random forest). Nothing to announce yet.

qm13, 2021-03-24:
Duplicate of #738.

closed 2021-03-24 by qm13