microsoft/Azure-Kinect-Sensor-SDK #1358: Use Multiple Azure Kinect depth sensors and its FPS

fshsu, 2020-09-18:
Is it possible to use multiple Azure Kinect depth sensors to capture a body at 30FPS. For example, the camera is on the left, right and front of the user,  and every depth view at 30FPS? Thanks.

qm13, 2020-09-21:
The Body Tracking SDK only supports images from a single camera. That said I am aware of some customers (including Microsoft internal teams) that are using multiple instances of body tracking with overlapping cameras and merging the skeletons taking into account the joint confidence. Running multiple cameras at 30fps requires understanding of the PCs USB controller architecture and bandwidth limitations. Running multiple instances of body tracking requires sufficient GPU (the Microsoft internal team is running 3 instances on a 2080).

fshsu, 2020-09-22:
Thanks the reply. We can do the Body Tracking on our development. However, We need the multiple depth cameras solution. Therefore, we would like to confirm the capture capability of Azure Kinect. Can the Azure Kinect get depth view at 30fps on every instance when using 3 instances?

jmachowinski, 2020-10-17:
Yes, we got up to 5 cameras working. 
Keep in mind, you may need additional USB host controllers.
Also it won't work on integrated graphics. 
We got 2 camera working with 30 fps using an integrated graphics chip of an i7-7700T.
For 5 camera we needed to switch to at least an 1050 TI graphics card.

sjdrc, 2020-10-21:
> Yes, we got up to 5 cameras working.
> Keep in mind, you may need additional USB host controllers.
> Also it won't work on integrated graphics.
> We got 2 camera working with 30 fps using an integrated graphics chip of an i7-7700T.
> For 5 camera we needed to switch to at least an 1050 TI graphics card.

How many cameras could you get running on a single USB3 host controller?

jmachowinski, 2020-10-28:
> How many cameras could you get running on a single USB3 host controller?
Depends
Q170 Chiphset : 2
Q470 Chipset : 4

closed 2020-11-30 by qm13